WPIRoboticsProjects/GRIP #94: Switch statements should have a default label

JLLeitschuh, 2015-11-06:
### [Codacy](https://www.codacy.com/app/WPIRoboticsProjects/GRIP/commit?cid=19129365) detected an issue:
#### Message: `Switch statements should have a default label`

    
#### Occurred on:
+ **Commit**:  ed3e9ee537ae38068da780ba4516df557d691d64
+ **File**:    [src/main/java/edu/wpi/grip/ui/util/ImageConverter.java](https://github.com/WPIRoboticsProjects/GRIP/blob/ed3e9ee537ae38068da780ba4516df557d691d64/src/main/java/edu/wpi/grip/ui/util/ImageConverter.java)
+ **LineNum**: [60](https://github.com/WPIRoboticsProjects/GRIP/blob/ed3e9ee537ae38068da780ba4516df557d691d64/src/main/java/edu/wpi/grip/ui/util/ImageConverter.java#L60)
+ **Code**:    `        switch (channels) {`

    
#### Currently on:
+ **Commit**:  c0df380eb03f3598af4e3ace2d56d58b1613c6f0
+ **File**:    [src/main/java/edu/wpi/grip/ui/util/ImageConverter.java](https://github.com/WPIRoboticsProjects/GRIP/blob/c0df380eb03f3598af4e3ace2d56d58b1613c6f0/src/main/java/edu/wpi/grip/ui/util/ImageConverter.java)
+ **LineNum**: [60](https://github.com/WPIRoboticsProjects/GRIP/blob/c0df380eb03f3598af4e3ace2d56d58b1613c6f0/src/main/java/edu/wpi/grip/ui/util/ImageConverter.java#L60)

ThomasJClark, 2015-11-06:
@JLLeitschuh, turn off this codacy thing

JLLeitschuh, 2015-11-06:
Did you resolve this issue?

ThomasJClark, 2015-11-06:
This isn't an issue

https://github.com/WPIRoboticsProjects/GRIP/blob/ed3e9ee537ae38068da780ba4516df557d691d64/src/main/java/edu/wpi/grip/ui/util/ImageConverter.java#L38

JLLeitschuh, 2015-11-06:
Assertions are turned off in the full version of the code.

JLLeitschuh, 2015-11-06:
Had I seen this I wouldn't have approved the PR.
That won't happen again thanks to codacity.

closed 2016-01-03 by JLLeitschuh