WPIRoboticsProjects/GRIP #126: Exception: cvGrabFrame() Error: Could not grab frame. (Has start() been called?)

bradamiller, 2015-11-11:
What were the actions performed prior to this error appearing?

## System Info:

Property Name | Property 
 ----- | -----
javafx.version | 8.0.60
java.runtime.name | Java(TM) SE Runtime Environment
java.vm.version | 25.60-b23
java.vm.vendor | Oracle Corporation
java.vm.name | Java HotSpot(TM) 64-Bit Server VM
java.runtime.version | 1.8.0_60-b27
java.awt.graphicsenv | sun.awt.CGraphicsEnvironment
javafx.runtime.version | 8.0.60-b27
os.name | Mac OS X
os.version | 10.11.1
os.arch | x86_64
file.encoding | UTF-8
java.vm.info | mixed mode
java.version | 1.8.0_60
sun.arch.data.model | 64
sun.cpu.endian | little

## Stack Trace:
```java
java.lang.IllegalStateException: Failed to grab image
	at edu.wpi.grip.core.sources.CameraSource.lambda$startVideo$23(CameraSource.java:110)
	at java.lang.Thread.run(Thread.java:745)
Caused by: org.bytedeco.javacv.FrameGrabber$Exception: cvGrabFrame() Error: Could not grab frame. (Has start() been called?)
	at org.bytedeco.javacv.OpenCVFrameGrabber.grab(OpenCVFrameGrabber.java:244)
	at edu.wpi.grip.core.sources.CameraSource.lambda$startVideo$23(CameraSource.java:108)
	... 1 more

```

JLLeitschuh, 2016-01-04:
I need more information. What type of camera source were you opening?

JoeTodes, 2016-01-15:
I'm getting an identical stack trace. We are using an Axis M1014. We can see that the camera is working by checking out the stream/setting page it serves up at 10.TE.AM.11.

In GRIP, we set the IP of the IP camera to this, an image initially appears in the GRIP preview, but it doesn't update new frames. Then when we deploy we get the above exception. After hitting Cancel, the deploy continues and we are able to run the file in headless mode on the RoboRio, but soon after it starts up we get:

SEVERE: Input must have a value to run this step

Followed by errors from each element in the pipeline saying that they require values.

JLLeitschuh, 2016-01-27:
This should be fixed by #392

AustinShalit, 2016-07-30:
This should have been fix in #392.  If you are still seeing this issue, please reopen this with a comment.

closed 2016-07-30 by AustinShalit