WPIRoboticsProjects/GRIP #134: GRIP should not prompt me to use webcams I don't have

nightpool, 2015-11-13:
Trying to use any webcam other then 0 when your system only has 1 webcam throws an exception. Ideally GRIP would enumerate the webcams on your system and only present ones that *won't* make it crash.

## Stack Trace:
```java
java.io.IOException: A problem occurred trying to start the frame grabber for Webcam 1
	at edu.wpi.grip.core.sources.CameraSource.startVideo(CameraSource.java:97)
	at edu.wpi.grip.core.sources.CameraSource.<init>(CameraSource.java:68)
	at edu.wpi.grip.core.sources.CameraSource.<init>(CameraSource.java:43)
	at edu.wpi.grip.ui.pipeline.AddSourceView.lambda$null$39(AddSourceView.java:94)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)
Caused by: org.bytedeco.javacv.FrameGrabber$Exception: cvCreateCameraCapture() Error: Could not create camera capture.
	at org.bytedeco.javacv.OpenCVFrameGrabber.start(OpenCVFrameGrabber.java:179)
	at edu.wpi.grip.core.sources.CameraSource.startVideo(CameraSource.java:95)
	... 6 more

```

JLLeitschuh, 2015-11-13:
Unfortunately, there doesn't seem to be any way to get this information out of the library we are using.
I agree that this is a feature that we need to figure out.

AustinShalit, 2016-07-11:
We could try to create the source and if it fails to start assume there is no camera attached.

SamCarlberg, 2016-07-11:
It will also crash if a webcam is present but another process is using it (at least on my macbook air) so that might not be the best way of doing it

JLLeitschuh, 2016-07-12:
We do this currently.
The crashing, if due to a segfault, we can't do anything about.

AustinShalit, 2016-08-02:
I have added a wontfix label to this.  It is important that GRIP gives access to cameras that do not exist for different environments.

closed 2016-08-02 by AustinShalit