WPIRoboticsProjects/GRIP #135: NullPointerException: Lost Connection to IP Webcam

nightpool, 2015-11-13:
After shutting down my IP webcam, GRIP threw a null pointer error. Ideally, GRIP would notice that it lost connection, and display the last frame it received as well as a "lost connection" notice somewhere in the UI. Ideal behavior would also include a UI control or intermittent pinging to restore connectivity when it becomes available. 

## Stack Trace:
```java
java.lang.NullPointerException
	at org.bytedeco.javacv.IPCameraFrameGrabber.grab(IPCameraFrameGrabber.java:121)
	at edu.wpi.grip.core.sources.CameraSource.lambda$startVideo$23(CameraSource.java:108)
	at java.lang.Thread.run(Thread.java:745)

```

JLLeitschuh, 2016-01-05:
@nightpool Can you try to reproduce this problem in the v0.3.0 release?

PaulaRudy, 2016-01-06:
Whoops, sorry about that.

PR #276 (2016-01-07): Fixes various problems CameraSource
 - The camera source could enter states where the the camera had crashed but
didn't properly stop itself and indicate this to the UI.

Closes #135
Related https://github.com/bytedeco/javacv/issues/299
commit 1e9e6f4d (2016-01-07): Fixes various problems CameraSource

 - The camera source could enter states where the the camera had crashed but
didn't properly stop itself and indicate this to the UI.

Closes #135
Related https://github.com/bytedeco/javacv/issues/299
closed 2016-01-07 by ThomasJClark