WPIRoboticsProjects/GRIP #142: Exception: cvCreateCameraCapture() Error: Could not create camera capture.

fauge7, 2015-11-13:
What were the actions performed prior to this error appearing?

Also opening a camera that does not exist causes an exception.

## System Info:

Property Name | Property 
 ----- | -----
javafx.version | 8.0.66
java.runtime.name | Java(TM) SE Runtime Environment
java.vm.version | 25.66-b17
java.vm.vendor | Oracle Corporation
java.vm.name | Java HotSpot(TM) 64-Bit Server VM
java.runtime.version | 1.8.0_66-b17
java.awt.graphicsenv | sun.awt.Win32GraphicsEnvironment
javafx.runtime.version | 8.0.66-b17
os.name | Windows 10
os.version | 10.0
os.arch | amd64
file.encoding | Cp1252
java.vm.info | mixed mode
java.version | 1.8.0_66
sun.arch.data.model | 64
sun.cpu.endian | little

## Stack Trace:
```java
java.io.IOException: A problem occurred trying to start the frame grabber for Webcam 1
	at edu.wpi.grip.core.sources.CameraSource.startVideo(CameraSource.java:97)
	at edu.wpi.grip.core.sources.CameraSource.<init>(CameraSource.java:68)
	at edu.wpi.grip.core.sources.CameraSource.<init>(CameraSource.java:43)
	at edu.wpi.grip.ui.pipeline.AddSourceView.lambda$null$39(AddSourceView.java:94)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)
Caused by: org.bytedeco.javacv.FrameGrabber$Exception: cvCreateCameraCapture() Error: Could not create camera capture.
	at org.bytedeco.javacv.OpenCVFrameGrabber.start(OpenCVFrameGrabber.java:179)
	at edu.wpi.grip.core.sources.CameraSource.startVideo(CameraSource.java:95)
	... 6 more

```

bobbysq, 2015-11-17:
Confirming.

closed 2015-11-30 by PaulaRudy

jojoguy10, 2015-12-08:
I'm still having issues with previewing the webcam image on Windows 10 64-bit

JLLeitschuh, 2015-12-08:
@jojoguy10 We don't have an official release out but Webcams have been fixed for windows in the newest version.
If you want to try out the latest development build you can download it here: https://ci.appveyor.com/project/JLLeitschuh/grip/build/artifacts

jojoguy10, 2015-12-08:
Thank you.  I'll just wait for the official release.

JLLeitschuh, 2015-12-10:
@jojoguy10 Its fixed in the most recent release. Give it a shot!

jojoguy10, 2015-12-10:
I just created a new issue with the error that came up

PR #166 (2015-11-30): Use videoInput for webcam capture on Windows
OpenCV's frame grabber framework seems to only work reliably for us on linux and osx, leading to crashes when using webcams on Windows.

Before this is merged, I'd like to get some testing done on a couple different computers and webcams.

closes #142
commit 64c9cd0e (2015-11-30): Use videoInput for webcam capture on Windows

OpenCV's frame grabber framework seems to only work reliably for us on
linux and osx, leading to crashes when using webcams on Windows.

closes #142