WPIRoboticsProjects/GRIP #145: Add and Add weighted only work with images of the same size.

PaulaRudy, 2015-11-14:
...At least in ubuntu. Is this something we want? Opencv allows for differently sized images, but the usage is complicated.

ThomasJClark, 2015-11-15:
I guess the most reasonable way to handle that would be to use the dimensions of the larger input as the dimensions of the output, and just add the smaller input to the top-left, like this:

[image]

Implementing that is pretty straightforward with OpenCV regions of interest.  This shouldn't have any significant overhead, since ROIs don't actually copy the regions they represent.

    // Assuming src2.cols() > src1.cols() and src2.rows() > src1.rows()

    // Set the output to the larger input
    src2.copyTo(dst);

    // Add the smaller input to the region in the upper-left corner of the output
    final Mat dstRoi = dst.apply(new Rect(new Point(0, 0), src1.size()));
    add(src1, dstRoi, dstRoi);

(The test code doesn't take into account the case where one input is taller but the other is wider, but I'm sure that wouldn't be too much more complex to handle.)

The real question is whether we want to do this in the generated code or not.  If not, then `add`, `addWeighted`, `bitwise_*`, and many similar operations would need to change to be hand-written.  IMO this is pretty basic, zero-cost functionality that should be included as part of the boilerplate in the generated OpenCV wrappers, and it would also eliminate a lot of the silly stack traces you get all the time. @JLLeitschuh?

ThomasJClark, 2015-11-15:
A simpler way to do this could also be to always use the dimensions of the first source as the dimensions of the output

PaulaRudy, 2015-11-15:
I believe this may be an issue with all operations that take two matrices? It also throws an error/exception  when you add the first image connection:

OpenCV Error: Sizes of input arguments do not match (The operation is neither 'array op array' (where arrays have the same size and the same number of channels), nor 'array op scalar', nor 'scalar op array') in arithm_op, file /bytedeco/javacpp-presets/opencv/cppbuild/linux-x86_64/opencv-3.0.0/modules/core/src/arithm.cpp, line 1987
java.lang.RuntimeException: /bytedeco/javacpp-presets/opencv/cppbuild/linux-x86_64/opencv-3.0.0/modules/core/src/arithm.cpp:1987: error: (-209) The operation is neither 'array op array' (where arrays have the same size and the same number of channels), nor 'array op scalar', nor 'scalar op array' in function arithm_op