WPIRoboticsProjects/GRIP #150: Mask is not working properly.

PaulaRudy, 2015-11-16:
[image]

ThomasJClark, 2015-11-16:
Could you post what steps you did to get this output (a screenshot would be fine) and the source images you used?

PaulaRudy, 2015-11-16:
Image or mask does not seem to matter, nor does adding sequence. Except for horizontal lines.
[image]
[image]
[image]
[image]

ThomasJClark, 2015-11-16:
OK.  Can I see what steps you did to get this output and the source images you used?

ThomasJClark, 2015-11-16:
It sounds like there are a lot of purely graphical defects that only happen on that one computer.  What video card do you have and which driver version are you using?

PaulaRudy, 2015-11-16:
I have an NVIDIA GT216 (GeForce GT 220) using the Nouveau display driver from xserver-xorg-video-nouveau. I am running Ubuntu Studio using the Xfce desktop environment (version 4.12). Ubuntu version is 15.04 (vivid vervet).

ThomasJClark, 2015-11-16:
Can you try using the proprietary nvidia driver? I just want to rule out graphics problems, since this doesn't happen in an Ubuntu vm for me

JLLeitschuh, 2015-12-05:
@PaulaRudy Can you upload the images you used to do this test?

ThomasJClark, 2015-12-05:
This is a non-issue.  The masks have lossy compression applied, so some pixels around the edges of the dark areas are not exactly 0.

It can be solved by using PNGs for masks.

closed 2015-12-05 by ThomasJClark