WPIRoboticsProjects/GRIP #161: Sources should have a start and stop method

JLLeitschuh, 2015-11-20:
Sources should not be started until either the user chooses to add it in the UI, the core triggers a start event (headless. This will ensure that sources that are loaded from a 

Starting a source:
- UI 
  - New sources (added by the UI) should be verified that they can start without a problem
     - These sources will start running after the creation UI closes
   - Sources loaded from a save file will be created stopped
     - These sources will start running after the user presses the "play" button
     - There will also be a "start all sources" button that will post a `StartAllSourcesEvent`
- Headless mode
  - Sources loaded from a save file will be created stopped
    - These sources will be initialized with a post `StartAllSourcesEvent`

ThomasJClark, 2015-11-21:
I don't think having a `StartAllSourcesEvent` has any real use. Why not just do a `StartSourceEvent` for every source you want to start? 

Users probably shouldn't sometimes have to press play and sometimes not. I think it would be confusing of a camera just works when you first add it, but not if you save and reload.

JLLeitschuh, 2015-11-21:
You shouldn't be able to create a source that isn't valid

ThomasJClark, 2015-11-21:
I agree. Was that in response to something I said, or a general statement?

JLLeitschuh, 2015-11-21:
Well you can't tell if a camera is invalid without starting it.

closed 2015-12-10 by JLLeitschuh