WPIRoboticsProjects/GRIP #163: Kinect Support

codetheweb, 2015-11-23:
Do you have any plans to package Kinect support with GRIP?  I'm not sure how it would work, since every platform implements drivers differently, but it could potentially be very useful.

JLLeitschuh, 2015-11-23:
How would you want to use GRIP with a Kinect? Are you looking to do things with the 3d matrix or just the 2D camera?

codetheweb, 2015-11-23:
I was thinking the depth stream.  For example, grouping objects with similar depths together or filtering by a certain depth.

JLLeitschuh, 2015-11-23:
If you are interested in opening a pull request to support this then we will take a look.
One other thing that GRIP will support that you may find useful is custom python scripts. This will allow you to write custom operations if the operations that we support don't work with the Kinect.
Supposedly the opencv library we are using will support the Kinect as is (as long as you plug in the right camera index when selecting your webcam).

JacisNonsense, 2016-03-13:
@JLLeitschuh mind assigning this to me? I'll attempt to resolve this at the season's conclusion.

SamCarlberg, 2016-10-31:
@JacisNonsense Have you got anywhere with this?

JacisNonsense, 2016-11-02:
Unfortunately not, I'm trying to think of a good way to load the Libfreenect Native Library while still maintaining platform compatibility.

JLLeitschuh, 2016-11-02:
Is there a problem in that the library isn't supported on all OS's or that there isn't a compiled binary for all OS's?

We have a toolchain you can use to compile for ARM linux if that's what you're missing.

JacisNonsense, 2016-11-02:
I'm aware of the toolchain. It is possible to package all the compatible binaries for libfreenect with the jar, but this would bloat the binary size and also require a partial rewrite of libfreenect's java wrapper.