WPIRoboticsProjects/GRIP #196: Dependency Injection Pattern Choice

JLLeitschuh, 2015-12-09:
# Dependency Injection

## Field Injection
 - Impossible to create instances of a class without using reflection
 - Reduces the amount of "boilerplate" code you have to write.
 - Harder to test

   > Field Injection
Guice injects fields with the @Inject annotation. This is the most concise injection, but the least testable.
   > \-[Guice wiki](https://github.com/google/guice/wiki/Injections)

 - Slower
   > Second issue is performance. A constructor call (direct or by reflection) is always faster than a bunch of reflection field assignments. Dependency injection frameworks must use reflection analysis to build dependency tree and create reflection constructors. This process causes additional performance hit.
  > \-[Maciej Chałapuk](http://programmers.stackexchange.com/a/300751)

 - Breaks the encapsulation paradigm

  > From the philosophical point of view, field injection breaks encapsulation, which is one of four fundamentals of object-oriented programming, which is the main paradigm of Java.
  > \-[Maciej Chałapuk](http://programmers.stackexchange.com/a/300751)

## Constructor Injection
 - Promotes smaller classes
   - If you start adding too many constructor fields it becomes a sign to refactor your code.
 - Promotes being able to define concrete state rules about a class.

   > The core of the problem here is that code you’ve written allows clients to create instances of the class in an invalid state. The whole purpose of a type is clients being able to rely on the invariants it enforces. It’s one of the reasons you use an EmailAddress type over a plain String to represent email adresses in your code: clients can be sure the instance they get is a valid email address as the value object enforces this constraint during construction. A String can potentially be anything, validated or not - how do you know?
    > \-[Oliver Gierke](http://olivergierke.de/2013/11/why-field-injection-is-evil/)

 - Provides a way to write tests without an injection framework

   > At least with constructor and setter injection, you have the option of not using the frameworks and/or reflection in your tests.
   > \- [Marjan Venema](http://programmers.stackexchange.com/a/300826)

ThomasJClark, 2015-12-09:
> * Harder to test

I don't think we currently test anything by passing mock objects into constructors, but if we ever decide to, for future reference, we can still create a `MockModule` and do field injection.  This shouldn't be a problem at all.

ThomasJClark, 2015-12-09:
> Considered 'Auto-Magic' by manny

> So i have always stayed away from field injection... There is a lot of what I consider automagic behind the scenes, and from my opinion I always prefer explicit over implicit.
- Scott Hepburn (Kronos Inc. Software Engineer, uses Spring)

This seems like an empty argument.  Everything computers do is "automagic".  Should we not use Java, because the freeing memory happens implicitly?  It's a tradeoff, sure, but having something that's really just useless boilerplate happen automatically is good.

ThomasJClark, 2015-12-09:
> Impossible to create instances of a class without using reflection

It's generally considered an antipattern to sometimes manually create classes and sometimes use dependency injection.  According to the Guice wiki, "Perhaps worst of all, any direct use of a constructor circumvents Guice's object instantiation." and will "introduce tighter coupling within a codebase"

ThomasJClark, 2015-12-09:
> Slower

We're writing a computer vision application.  Constructing the several dozen objects we use in the application will take like 0.00001% of the time compare to the CV processing, regardless of how we do it

JLLeitschuh, 2015-12-09:
> Perhaps worst of all, any direct use of a constructor circumvents Guice's object instantiation.

Actually, this is that full quote is:

> Perhaps worst of all, any direct use of a constructor circumvents Guice's object instantiation.

 > As a correction, simply limit the visibility of both your implementation classes, and their constructors. Typically package private is preferred for both, as this facilitates:
- binding the class within a Module in the same package
- **unit testing the class through means of direct instantiation**

Unit testing should not use the dependency injection framework.

closed 2015-12-09 by ThomasJClark