WPIRoboticsProjects/GRIP #199: ArithmeticException: / by zero

jojoguy10, 2015-12-10:
What were the actions performed prior to this error appearing?

I added a webcam.  I'm using a USB webcam (Logitech 9000 Pro) on Windows 10 64-bit

Message: Camera Frame Grabber Thread crashed with uncaught exception

## System Info:

Property Name | Property 
 ----- | -----
javafx.version | 8.0.66
java.runtime.name | Java(TM) SE Runtime Environment
java.vm.version | 25.66-b17
java.vm.vendor | Oracle Corporation
java.vm.name | Java HotSpot(TM) 64-Bit Server VM
java.runtime.version | 1.8.0_66-b17
java.awt.graphicsenv | sun.awt.Win32GraphicsEnvironment
javafx.runtime.version | 8.0.66-b17
os.name | Windows 10
os.version | 10.0
os.arch | amd64
file.encoding | Cp1252
java.vm.info | mixed mode
java.version | 1.8.0_66
sun.arch.data.model | 64
sun.cpu.endian | little

## Stack Trace:
```java
java.lang.ArithmeticException: / by zero
	at edu.wpi.grip.core.sources.CameraSource.lambda$start$18(CameraSource.java:158)
	at java.lang.Thread.run(Thread.java:745)

```

ThomasJClark, 2015-12-10:
Probably because of [this](https://github.com/WPIRoboticsProjects/GRIP/blob/5db3adf16f606d22b7c1ca33324c91c536cd3c9a/core/src/main/java/edu/wpi/grip/core/sources/CameraSource.java#L157)

    long thisMoment = System.currentTimeMillis();
    frameRateOutputSocket.setValue(1000 / (thisMoment - lastFrame));

JLLeitschuh, 2015-12-10:
Thanks, I'll fix this. Does this happen every time or only sometimes?

JLLeitschuh, 2015-12-10:
@ThomasJClark does the Windows frame grabber block like the other ones do? That may have caused this.

jojoguy10, 2015-12-10:
It happened to me every time.  I couldn't get my webcam to work at all.

closed 2015-12-11 by PaulaRudy

JLLeitschuh, 2015-12-15:
@jojoguy10 We have a patch up that should fix your problem. Thanks for reporting it!
https://github.com/WPIRoboticsProjects/GRIP/releases

PR #201 (2015-12-11): Fix Divide by Zero Error in CameraSource
Closes #199
commit 7ee0f71c (2015-12-11): Fix Divide by Zero Error in CameraSource

Closes #199