WPIRoboticsProjects/GRIP #200: Get this on startup

bradamiller, 2015-12-10:
Is this OK?

:ui:run
ID	Type	Value
Dec 10, 2015 9:50:18 AM com.sun.javafx.css.StyleConverterImpl getInstance
SEVERE: StyleConverterImpl : converter Class is null for : com.sun.javafx.scene.layout.region.CornerRadiiConverter
Dec 10, 2015 9:50:18 AM com.sun.javafx.css.StyleConverterImpl readBinary
SEVERE: could not deserialize com.sun.javafx.scene.layout.region.CornerRadiiConverter
could not deserialize com.sun.javafx.scene.layout.region.CornerRadiiConverter
> Building 90% > :ui:run

ThomasJClark, 2015-12-10:
Yep, that's from controlsfx. It doesn't seem to really be a problem.

closed 2015-12-10 by ThomasJClark

JLLeitschuh, 2015-12-10:
Congrats on the 200th issue? Hahah!
:fireworks: :clap: :fireworks: