WPIRoboticsProjects/GRIP #203: The RoboRIO Deployment Problem

JLLeitschuh, 2015-12-10:
# Sources
1. Having multiple clients try to access the frames of a webcam at the same time can create problems (right??) so we have to ensure that when GRIP deploys we have to stop the local instances sources.
2. Currently it is impossible to create a source that is not valid on your machine. This source may, however, due to networking or webcams being plugged into different ports, be valid on the roboRIO, how do we handle this?

# Sinks
If we have a sink that is publishing after we deploy GRIP over to the robot then there will be two network tables nodes publishing to the same values. As such, we should make it so that when when we push the jar over to the robot it should should suspend camera access on the local machine.

virtuald, 2015-12-28:
FWIW, if you install mjpg-streamer it can send streams to multiple clients at the same time -- so access the camera through that, instead of directly talking to the webcam. It's pretty light too.

JLLeitschuh, 2016-01-02:
@ThomasJClark thoughts?

JLLeitschuh, 2016-01-09:
This was fixed by @PaulaRudy #304

closed 2016-01-09 by JLLeitschuh