WPIRoboticsProjects/GRIP #214: Graphical errors in GUI

jojoguy10, 2015-12-15:
When the app is in fullscreen mode in Windows 10 64-bit, white boxes cover the options as I hover over them.
[image]
It only happens when the app is fullscreen.  The issue fixes itself after I switch to window mode.

JLLeitschuh, 2015-12-15:
Duplicate of #193
@ThomasJClark Do you have any idea how to fix this

PaulaRudy, 2015-12-15:
@jojoguy10 Thanks for your report! This is a known bug at this point, and we'll get working on this soon.

PaulaRudy, 2016-01-05:
I've seen this before but it appears to not be duplicatable in Linux. @JLLeitschuh, would you like to take a stab at this? I've got a few leads: It *could* possibly be related to hardware graphics acceleration (PRISM) in JavaFx, and we could *possibly* bypass this by either disabling PRISM or by forcing JavaFX to re-render (perhaps by resizing the column width? But, of course, this is a work around and does not fix the underlying issue).

JLLeitschuh, 2016-01-09:
This can not be reproduced in the current version of grip. Are you still seeing this bug @jojoguy10 ?

jojoguy10, 2016-01-12:
I have note been seeing this bug in the latest version of GRIP (ver1.0.0rc4)

JLLeitschuh, 2016-01-12:
I have no idea what fixed this but cool!!

closed 2016-01-12 by JLLeitschuh