WPIRoboticsProjects/GRIP #220: NullPointerException: null

yuryws, 2015-12-24:
What were the actions performed prior to this error appearing?
Axis M1011 camera was added

Message: Camera Frame Grabber Thread crashed with uncaught exception

## System Info:

Property Name | Property 
 ----- | -----
javafx.version | 8.0.66
java.runtime.name | Java(TM) SE Runtime Environment
java.vm.version | 25.66-b17
java.vm.vendor | Oracle Corporation
java.vm.name | Java HotSpot(TM) 64-Bit Server VM
java.runtime.version | 1.8.0_66-b17
java.awt.graphicsenv | sun.awt.Win32GraphicsEnvironment
javafx.runtime.version | 8.0.66-b17
os.name | Windows 7
os.version | 6.1
os.arch | amd64
file.encoding | Cp1252
java.vm.info | mixed mode
java.version | 1.8.0_66
sun.arch.data.model | 64
sun.cpu.endian | little

## Stack Trace:
```java
java.lang.NullPointerException
	at edu.wpi.grip.core.sources.IPCameraFrameGrabber.grab(IPCameraFrameGrabber.java:129)
	at edu.wpi.grip.core.sources.CameraSource.lambda$start$18(CameraSource.java:144)
	at java.lang.Thread.run(Thread.java:745)

```

ThomasJClark, 2015-12-24:
What did you put for the URL?

yuryws, 2015-12-24:
http://10.36.24.90

This is the IP address assigned to the camera (Axis M1011). This was tested using a browser and works fine.

ThomasJClark, 2015-12-24:
Try using http://10.36.24.90/mjpg/video.mjpg

JLLeitschuh, 2016-01-03:
Closed by #215

closed 2016-01-03 by JLLeitschuh