WPIRoboticsProjects/GRIP #225: ArrayIndexOutOfBoundsException

dangyogi, 2015-12-29:
What were the actions performed prior to this error appearing?

I had a webcam tied to an HSV Threshold and was viewing both in the top half of the screen.

The information below was captured by clicking the "Copy to Clipboard" button on the exception window and then pasting here.

Message: The UI Thread threw an uncaught exception

## System Info:

Property Name | Property 
 ----- | -----
javafx.version | 8.0.66
java.runtime.name | Java(TM) SE Runtime Environment
java.vm.version | 25.66-b17
java.vm.vendor | Oracle Corporation
java.vm.name | Java HotSpot(TM) 64-Bit Server VM
java.runtime.version | 1.8.0_66-b17
java.awt.graphicsenv | sun.awt.X11GraphicsEnvironment
javafx.runtime.version | 8.0.66-b17
os.name | Linux
os.version | 3.13.0-74-generic
os.arch | amd64
file.encoding | UTF-8
java.vm.info | mixed mode
java.version | 1.8.0_66
sun.arch.data.model | 64
sun.cpu.endian | little

## Stack Trace:
```java
java.lang.ArrayIndexOutOfBoundsException

```

JLLeitschuh, 2015-12-30:
Thanks.
What was the action you performed immediately preceding the exception appearing?
This may be related to #155

dangyogi, 2015-12-30:
Sorry, I don't really remember.  I'm just starting out with GRIP, so it was
pretty simple.  I had one webcam with the "eye" thing opened on top, and
that was linked to an HSV (I think, might have been HSL or RGB, but I'm
pretty sure HSV) threshold with its "eye" thing also open on top and I was
playing with the HSV threshold settings.  So two objects with two displays
on top.

But I'm kind of thinking too that it *may* have happened out of the blue
(not in response to me changing a setting).  "out of the blue" means that
the two "eye" views are open on top and the system is dynamically
processing images, but that I wasn't touching anything.

The webcam is a built-in laptop camera with a frame rate of ~30.  Not sure
what the resolution is.

I was surprised that the "Copy Error Info" button (or whatever it was
called) on the exception window didn't capture a stack backtrace to show
you where in the code the error happened.  Is there something that I can
set to turn on a backtrace?  (I'm not a java developer).

-Bruce

JLLeitschuh, 2015-12-30:
Yea, I'm confused as well. I've seen errors like that before and I'm not sure why the stack trace got lost.
We aren't intentionally compiling in such a way that we remove line numbers from the program.

JLLeitschuh, 2015-12-30:
Well, I may have a fix for the line number thing. This doesn't really fix your problem but it will make it easier for us to debug stuff in the future.

JLLeitschuh, 2015-12-30:
Unfortunately there's no way to fix this exact problem because we don't have a stack trace. However, you did expose another problem with how the JIT handles (eats) stack traces, so thank you. #227 should fix this problem so we don't have more disappearing stack traces.
This should be resolved in the next release.
Thanks for your help!!!

JLLeitschuh, 2016-01-03:
I'm going to take a gander that this was resolved by #231.
If you have this problem again with the newest release [v0.3.0](https://github.com/WPIRoboticsProjects/GRIP/releases/tag/v0.3.0) or higher.

closed 2016-01-03 by JLLeitschuh

dangyogi, 2016-01-03:
OK, thanks for looking at this!