WPIRoboticsProjects/GRIP #234: NoSuchElementException when previewing images

ThomasJClark, 2016-01-03:
This is on the latest master (8ad6c1a)

When I add a source and threshold as below, sometimes (not every time) I get the stack trace at the end of this post.  This started happening as soon as @JLLeitschuh merged the dependency injection PR.

[image]

Also, the filled-in socket handle in the source is no longer filled in.  I'm not sure if this is related.

```
java.util.NoSuchElementException: No value present
	at java.util.Optional.get(Optional.java:135)
	at edu.wpi.grip.ui.preview.PreviewsController.lambda$getIndexInPreviewsOfAStepSocket$78(PreviewsController.java:222)
	at edu.wpi.grip.ui.preview.PreviewsController$$Lambda$442/1454995922.test(Unknown Source)
	at java.util.stream.ReferencePipeline$2$1.accept(ReferencePipeline.java:174)
	at java.util.ArrayList$ArrayListSpliterator.forEachRemaining(ArrayList.java:1374)
	at java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:512)
	at java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:502)
	at java.util.stream.ReduceOps$ReduceOp.evaluateSequential(ReduceOps.java:708)
	at java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:234)
	at java.util.stream.LongPipeline.reduce(LongPipeline.java:438)
	at java.util.stream.LongPipeline.sum(LongPipeline.java:396)
	at java.util.stream.ReferencePipeline.count(ReferencePipeline.java:526)
	at edu.wpi.grip.ui.preview.PreviewsController.getIndexInPreviewsOfAStepSocket(PreviewsController.java:222)
	at edu.wpi.grip.ui.preview.PreviewsController.lambda$onSocketPreviewChanged$77(PreviewsController.java:154)
	at edu.wpi.grip.ui.preview.PreviewsController$$Lambda$405/1715843290.run(Unknown Source)
	at edu.wpi.grip.ui.util.GRIPPlatform.runAsSoonAsPossible(GRIPPlatform.java:63)
	at edu.wpi.grip.ui.preview.PreviewsController.onSocketPreviewChanged(PreviewsController.java:143)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:497)
	at com.google.common.eventbus.EventSubscriber.handleEvent(EventSubscriber.java:74)
	at com.google.common.eventbus.SynchronizedEventSubscriber.handleEvent(SynchronizedEventSubscriber.java:47)
	at com.google.common.eventbus.EventBus.dispatch(EventBus.java:322)
	at com.google.common.eventbus.EventBus.dispatchQueuedEvents(EventBus.java:304)
	at com.google.common.eventbus.EventBus.post(EventBus.java:275)
	at edu.wpi.grip.core.OutputSocket.setPreviewed(OutputSocket.java:53)
	at edu.wpi.grip.ui.pipeline.OutputSocketController.lambda$initialize$33(OutputSocketController.java:73)
	at edu.wpi.grip.ui.pipeline.OutputSocketController$$Lambda$402/1971128061.invalidated(Unknown Source)
	at com.sun.javafx.binding.ExpressionHelper$SingleInvalidation.fireValueChangedEvent(ExpressionHelper.java:137)
	at com.sun.javafx.binding.ExpressionHelper.fireValueChangedEvent(ExpressionHelper.java:81)
	at javafx.beans.property.BooleanPropertyBase.fireValueChangedEvent(BooleanPropertyBase.java:103)
	at javafx.beans.property.BooleanPropertyBase.markInvalid(BooleanPropertyBase.java:110)
	at javafx.beans.property.BooleanPropertyBase.set(BooleanPropertyBase.java:144)
	at javafx.scene.control.ToggleButton.setSelected(ToggleButton.java:149)
	at javafx.scene.control.ToggleButton.fire(ToggleButton.java:255)
	at com.sun.javafx.scene.control.behavior.ButtonBehavior.mouseReleased(ButtonBehavior.java:182)
	at com.sun.javafx.scene.control.skin.BehaviorSkinBase$1.handle(BehaviorSkinBase.java:96)
	at coavafx.scene.control.skin.BehaviorSkinBase$1.handle(BehaviorSkinBase.java:89)
	at com.sun.javafx.event.CompositeEventHandler$NormalEventHandlerRecord.handleBubblingEvent(CompositeEventHandler.java:218)
	at com.sun.javafx.event.CompositeEventHandler.dispatchBubblingEvent(CompositeEventHandler.java:80)
	at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:238)
	at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:191)
	at com.sun.javafx.event.CompositeEventDispatcher.dispatchBubblingEvent(CompositeEventDispatcher.java:59)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:58)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.EventUtil.fireEventImpl(EventUtil.java:74)
	at com.sun.javafx.event.EventUtil.fireEvent(EventUtil.java:54)
	at javafx.event.Event.fireEvent(Event.java:198)
	at javafx.scene.Scene$MouseHandler.process(Scene.java:3758)
	at javafx.scene.Scene$MouseHandler.access$1500(Scene.java:3486)
	at javafx.scene.Scene.ocessMouseEvent(Scene.java:1762)
	at javafx.scene.Scene$ScenePeerListener.mouseEvent(Scene.java:2495)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler$MouseEventNotification.run(GlassViewEventHandler.java:350)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler$MouseEventNotification.run(GlassViewEventHandler.java:275)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler.lambda$handleMouseEvent$350(GlassViewEventHandler.java:385)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler$$Lambda$341/882213246.get(Unknown Source)
	at com.sun.javafx.tk.quantum.QuantumToolkit.runWithoutRenderLock(QuantumToolkit.java:404)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler.handleMouseEvent(GlassViewEventHandler.java:384)
	at com.sun.glass.ui.View.handleMouseEvent(View.java:555)
	at com.sun.glass.ui.View.notifyMouse(View.java:927)
	at com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
	at com.sun.glass.ui.gtk.GtkApplication.lambda$null$48(GtkApplication.java:139)
	at com.sun.glass.ui.gtk.GtkApplication$$Lambda$41/1327763628.run(Unknown Source)
	at java.lang.Thread.run(Thread.java:745)
```

JLLeitschuh, 2016-01-03:
Reproducible.

JLLeitschuh, 2016-01-03:
Found the problem and have a fix

PR #237 (2016-01-03): Fixes NoSuchElement when previewing a Source then a Step
Closes #234
commit c551d6d6 (2016-01-03): Fixes NoSuchElement when previewing a Source then a Step

Closes #234
closed 2016-01-03 by JLLeitschuh