WPIRoboticsProjects/GRIP #245: Should not be able to save a pipeline with errors (or at least shouldn't be able to deploy)

JLLeitschuh, 2016-01-04:
If a step has a problem then you should not be able to deploy to the robot. This will prevent confusing "why didn't this work" problems. It will prevent you from shooting yourself in the foot.

PaulaRudy, 2016-01-06:
I dunno about not saving- I think that would be frustrating. Maybe pop up a warning message? ("This pipeline has errors that are not addressed. Are you sure you want to deploy?" )

PaulaRudy, 2016-01-06:
It's not really a bug though IMO.

ThomasJClark, 2016-01-06:
A pipeline could still be work if it has some steps that aren't connected to anything.  We shouldn't stop people from deploying a perfectly working project, and we DEFINITELY should not prevent them from saving one

PaulaRudy, 2016-01-06:
I'm with @ThomasJClark  on this one. I think maybe a popup warning for deployment might be a good idea, but I'm worried about it getting annoying.

AustinShalit, 2016-07-30:
I am going to close this issue.  I agree that a pipeline could still work when deployed even if there are problems when it is running on the computer.

closed 2016-07-30 by AustinShalit