WPIRoboticsProjects/GRIP #252: Operations feed back to previous step

JLLeitschuh, 2016-01-04:
I loaded this this save file:
```xml
<grip:Pipeline>
  <sources>
    <grip:Camera>
      <property name="deviceNumber" value="0"/>
    </grip:Camera>
  </sources>
  <steps>
    <grip:Step name="resize">
      <grip:Input step="0" socket="0"/>
      <grip:Input step="0" socket="1"/>
      <grip:Input step="0" socket="2">
        <value>0.3</value>
      </grip:Input>
      <grip:Input step="0" socket="3">
        <value>0.3</value>
      </grip:Input>
      <grip:Input step="0" socket="4">
        <value>INTER_LINEAR</value>
      </grip:Input>
      <grip:Output step="0" socket="0" previewed="true"/>
    </grip:Step>
    <grip:Step name="Desaturate">
      <grip:Input step="1" socket="0"/>
      <grip:Output step="1" socket="0" previewed="true"/>
    </grip:Step>
    <grip:Step name="Find Lines">
      <grip:Input step="2" socket="0"/>
      <grip:Output step="2" socket="0" previewed="true"/>
    </grip:Step>
  </steps>
  <connections>
    <grip:Connection>
      <grip:Output step="0" socket="0" previewed="true"/>
      <grip:Input step="1" socket="0"/>
    </grip:Connection>
    <grip:Connection>
      <grip:Output step="1" socket="0" previewed="true"/>
      <grip:Input step="2" socket="0"/>
    </grip:Connection>
    <grip:Connection>
      <grip:Output source="0" socket="0" previewed="false"/>
      <grip:Input step="0" socket="0"/>
    </grip:Connection>
  </connections>
</grip:Pipeline>
```
And the pipeline looked like this:
[image]

Notice:
 - The steps don't have their previewed buttons toggled
 - The first step that is just a simple resize is actually being desaturated as well (this issue)

The pipeline should look like this:
[image]

PaulaRudy, 2016-01-06:
Seems to be associated specifically with the desaturate step. The preview windows being open is an artifact of them being open at the time the file was saved. Here's the output of the spectacular error I managed to trrigger
[GRIP.txt](https://github.com/WPIRoboticsProjects/GRIP/files/80495/GRIP.txt)

PaulaRudy, 2016-01-08:
Tell me what you think? It's not the most elegant solution, but it works.

JLLeitschuh, 2016-01-09:
Yea, this is not the solution we should be going with. There is a much more elegant change that should only require one line of code to be changed plus adding some ignores to some tests.

PaulaRudy, 2016-01-10:
FYI to all interested: Apparently my solution was premature. Still working on this.

PaulaRudy, 2016-01-10:
Are we using hash tables in the serialization process? I'm thinking this might be a hash collision that results in two references to the same memory space.

ThomasJClark, 2016-01-10:
No, I don't think so. 

Can you check if the output of the step is actually incorrect, or only the preview? Because it could just be a bug in `PreviewsController`

PaulaRudy, 2016-01-10:
>Can you check if the output of the step is actually incorrect, or only the preview?

Ooo...good idea!

PaulaRudy, 2016-01-11:
Looks like the error is present within the pipeline too. The preview entries point to the right step, but both the entries in the list of previewed sockets AND the entries in the pipeline point to the same mat for both steps.
[image]

JLLeitschuh, 2016-01-14:
That is really good information. Thanks!

JLLeitschuh, 2016-01-15:
Thanks @PaulaRudy that the info I needed to solve the problem.
We really should have just looked at the desaturate operation and the problem would have been right there.
I thought that it was caused by some greater design problem but it wasn't it was just in the desaturate operation.

PR #362 (2016-01-15): Fixes Desaturate Feeding back into the previous step
This was caused by not copying the input to the output
and just simply assigning the value of the input to the
output.

Closes #252

I actually had a comment on that line when this code was introduced but the comment was never resolved.
https://github.com/WPIRoboticsProjects/GRIP/pull/63/files#r43678331
Any way this can be prevented in the future?
commit 0c3dc290 (2016-01-15): Fixes Desaturate Feeding back into the previous step

This was caused by not copying the input to the output
and just simply assigning the value of the input to the
output.

Closes #252
closed 2016-01-15 by ThomasJClark