WPIRoboticsProjects/GRIP #256: The deploy icon is tiny

ThomasJClark, 2016-01-04:
The deploy icon in the menu is tiny my HiDPI screen.  I'm not sure if we can get controlsfx glyphs to scale correctly, so it might be a good idea to just use a png icon like the rest of the menu items.

JLLeitschuh, 2016-01-05:
Can you open an issue with controlsfx?

ThomasJClark, 2016-01-05:
I don't think this is a bug in ControlsFX.  Normally in JavaFX graphics don't inherit CSS rules

JLLeitschuh, 2016-01-05:
We can set the size here.
Lets just add something to the DPI file to do this automatically:
http://controlsfx.bitbucket.org/org/controlsfx/glyphfont/GlyphFont.html

ThomasJClark, 2016-01-05:
Can you figure out how to do that in FXML?

JLLeitschuh, 2016-01-05:
No, can you. I already have enough issues tagged with my name on them right now.

ThomasJClark, 2016-01-05:
I think it should be an FRC logo rather than an upload logo.  Once NT is merged it (which adds an FRC logo to the resources) I'll change it to do that

AustinShalit, 2016-01-05:
Is this issue related to the picture attached?
[image]

ThomasJClark, 2016-01-05:
> Is this issue related to the picture attached?

I don't think so, but that's an interesting problem I haven't seen before.  Mind opening up a new issue?

PaulaRudy, 2016-01-06:
@JLLeitschuh does #279 address this as well?

ThomasJClark, 2016-01-06:
No, but #239 does because it just sets deploying to use the same icon as
NetworkTables

commit 8237c97c (2016-01-06): Use FRC logo for "FRC Deploy"

Might as well, since the associated PR adds an FRC icon to the project
resources

Fixes #256
closed 2016-01-06 by ThomasJClark