WPIRoboticsProjects/GRIP #258: Deploy should use the team number project setting

ThomasJClark, 2016-01-04:
From #239 

So, the entire right side of the window should not really be there, since we can get the team number from the setting (for the advanced option, we're going add a manual hostname setting anyways to the project settings)

JLLeitschuh, 2016-01-05:
Actually, that's not true. You may want to deploy to a specific address.
The team number should default to the one from the project settings.

What if I want to publish to a network tables server running on my local computer but I'm deploying to a remote device.

ThomasJClark, 2016-01-05:
I disagree.  I am confident that literally 100% of teams will deploy to the same address that they use for a networktables server, which is always on the robot.  They definitely also won't use two different team numbers.

It's possible that the robot might be at a different address besides the automatic mDNS one, but that can be solved with a manual hostname setting.

JLLeitschuh, 2016-01-05:
@AustinShalit You want to weigh in on this one?

I want to deploy to the robot but use a table viewer on my local machine running a server.

bradamiller, 2016-01-05:
Why not leave it in for now? It will definitely help us with testing, then we don't need to have a program running on the roboRIO and can still test GRIP. It's for the same reason that the OutlineViewer can run as a client or a server, it makes it easy to test other things without the robot running code.

JLLeitschuh, 2016-01-05:
So removing the UI is a *wontfix* but I will make it so that the UI uses the project defaults.

ThomasJClark, 2016-01-05:
> I want to deploy to the robot but use a table viewer on my local machine running a server.

This is why the table viewer has an option to run as a client

> Why not leave it in for now? It will definitely help us with testing, then we don't need to have a program running on the roboRIO and can still test GRIP. It's for the same reason that the [...]

This doesn't prevent us from running a networktables server locally for testing.  NetworkTables defaults to localhost if it can't connect to the server

> So removing the UI is a wontfix but I will make it so that the UI uses the project as defaults.

If there's really another option we want people to be able to change, put it in a project setting.

AustinShalit, 2016-01-05:
I have never seen a case in competition where a team was running the NetworkTables server on anything other than the RoboRIO.

JacisNonsense, 2016-01-05:
It would be nice to have an option for custom IP addresses, as some school networks may prevent mDNS discovery. Additionally, for those teams using co-processors, this option could be immensely useful.

AustinShalit, 2016-01-05:
I agree with @JacisNonsense

ThomasJClark, 2016-01-05:
I think that's a good plan.  Basically, remove this configuration from the deploy panel, but have the project settings have an option for custom address.

JLLeitschuh, 2016-01-05:
The address you deploy to may be different from the address your network tables server is located though. Deploy location should not be a project setting.

ThomasJClark, 2016-01-05:
> The address you deploy to may be different from the address your network tables server is located though.

I agree

> Deploy location should not be a project setting.

I disagree.  That would mean you have to re-enter it every time, which would be super annoying if you're deploying over and over again.

JacisNonsense, 2016-01-05:
>That would mean you have to re-enter it every time, which would be super annoying if you're deploying over and over again.

I agree

JLLeitschuh, 2016-01-05:
Fine but the deploy to robot address should be configurable from the deployment screen. Otherwise it will be a hassle if you want to change it for one deployment to an alternate test roborio.

So whatever you set in your project settings will be the default fields but you can change it in the deployment screen if you want to.

ThomasJClark, 2016-01-05:
> Fine but the deploy to robot address should be configurable from the deployment screen. Otherwise it will be a hassle if you want to change it for one deployment to an alternate test roborio.

Nah, just read it from the project settings.   I don't think FRC teams would ever frequently switch between two roboRIOs with different addresses.

AustinShalit, 2016-01-05:
> I don't think FRC teams would ever frequently switch between two roboRIOs with different addresses.

If you mean different IP addresses, my team builds 2 robots every year.  One is team 1836.  The other is team 9836.  We do this so we can have our drive team practice with other robots on the field.  Typically we use one robot as a development platform (can errors) and the other as a production (even if mechanisms are disabled -- never errors).

bradamiller, 2016-01-05:
Yeah, that's pretty standard for teams with multiple robots, they each get a different team number. That way they could both be operated at the same time. 

Brad

JLLeitschuh, 2016-01-05:
This is really valuable feedback. Thank you everyone for sharing your use case.

PaulaRudy, 2016-01-06:
Is this resolved then? Can we close this?

ThomasJClark, 2016-01-06:
> Is this resolved then? Can we close this?

no

JLLeitschuh, 2016-01-06:
This will be resolved and closed by a PR hopefully in the next day or so.

closed 2016-01-08 by JLLeitschuh