WPIRoboticsProjects/GRIP #260: There shouldn't be separate standard output and error windows in the deploy dialog

ThomasJClark, 2016-01-04:
I'm not sure if we should even print anything to stdout, but if we do, it would definitely be useful for both streams to be in the same window.  Otherwise, you can't tell what order things happened in.

[image]

JLLeitschuh, 2016-01-04:
There are two separate streams. You know how to merge them gracefully?
I certainly don't.

ThomasJClark, 2016-01-04:
Yeah, it should be as easy as providing the same `OutputStream` for both the `stdout` and `stderr` arguments of `Shell::exec()`

JLLeitschuh, 2016-01-05:
Won't that mean that we have no idea what character came from what stream?
That seems like a threading problem in the making.

ThomasJClark, 2016-01-05:
I don't think users care what came from what stream.  Preserving ordering is probably more important.

This is also the behavior that just about every other program has.  For example, IntelliJ shows both streams in one console when you run a program.

JLLeitschuh, 2016-01-05:
Character order matters.
You could end up with two lines merged into one if you don't.
```
STyshee omuertporut cjousuld lo hapok liedke this otherwise
```

ThomasJClark, 2016-01-05:
I'd rather that happen than we lose information about ordering.  That's the default behavior in every terminal emulator and IDE.  I also think we can mostly avoid that issue anyways by basically always using stderr (since we should only print things with the logging framework or when an exception happens)

JLLeitschuh, 2016-01-05:
This will not be fixed in time for v1.0.0

PaulaRudy, 2016-01-06:
You could always append a "stdout:" or "err:" to each output.

AustinShalit, 2016-01-06:
Or color the error output red

ThomasJClark, 2016-01-06:
> This is a won't fix at least until after v1.0.0
> We can discuss this more then

Why?  This is a super easy fix

PaulaRudy, 2016-01-06:
Well, if it works as is, there's no reason to make it priority right now, especially considering that there's much more critical things to worry about. But I do agree with you that this shouldn't be that difficult to fix.

JLLeitschuh, 2016-01-06:
How do you differentiate between each character coming in in the stream?
I could get a character flushed by the output stream and the error stream at the same time.

JLLeitschuh, 2016-01-27:
This was resolved by #386

closed 2016-01-27 by JLLeitschuh