WPIRoboticsProjects/GRIP #261: Code style

ThomasJClark, 2016-01-04:
We don't have a consistent code style defined.  It's kind of mostly been "whatever IntelliJ does by default, except when we forget to format or decide to do something different".

At some point, I'm gonna try to get the [checkstyle](https://docs.gradle.org/current/userguide/checkstyle_plugin.html) plugin integrated into the build.

JLLeitschuh, 2016-01-04:
You have to make sure whatever style sheet we use works with PMD.

ThomasJClark, 2016-01-04:
?

JLLeitschuh, 2016-01-04:
Codacity uses PMD under the hood.
For example, having annotations on the same line as the field declaration may break PMD.

ThomasJClark, 2016-01-04:
What do you mean "break PMD"?  Will it not work at all?

(btw, I don't know about you, but I've basically stopped looking at codacity reports, so I wouldn't miss it)

AustinShalit, 2016-01-05:
I am going to take a stab at setting this up.

AustinShalit, 2016-07-29:
We have added both Checkstyle and PMD to the build.

closed 2016-07-29 by AustinShalit