WPIRoboticsProjects/GRIP #262: Deploy menu item should probably be under "tools"

ThomasJClark, 2016-01-04:
I doubt we'll ever have more than one deploy option (FRC is the only one that we can do automatically since all of the FRC systems are the same).  So, "FRC Deploy" should probably be a single option under "Tools"

bradamiller, 2016-01-04:
I can see doing deployments to other devices like the Jetson or Raspberry PI that I have in my office. I’m not sure what’s involved, but I suspect that it’s very similar to the roboRIO. That would be very cool. 

Brad

JLLeitschuh, 2016-01-04:
I agree with brad on this one.

ThomasJClark, 2016-01-04:
Good point.  I think it's probably best under "tools", but I totally agree that it would cool to have a "deploy to raspberry pi" tool.

JLLeitschuh, 2016-01-04:
I think a deploy menu makes more sense.

ThomasJClark, 2016-01-04:
Why?

AustinShalit, 2016-01-04:
Deploy menu is going to be a high use item and should be not be hidden beneath menus.  Imagine if every time you wanted to compile in an IDE you had to go hunt for the compile button...

JLLeitschuh, 2016-01-04:
I agree with @AustinShalit on this one.

JLLeitschuh, 2016-01-05:
Can this be closed? I think you've been outvoted on this one @ThomasJClark sorry.

ThomasJClark, 2016-01-05:
> Can this be closed? I think you've been outvoted on this one @ThomasJClark sorry.

We currently have three ideas stated:

1. FRC Deploy should be under "tools"
2. FRC Deploy should be under "deploy"
3. FRC Deploy should be a button not in a menu

Each of these ideas was only proposed once, although I think you supported both 2 and 3

Which one are you saying wins?

JLLeitschuh, 2016-01-05:
2

PaulaRudy, 2016-01-06:
Has this been resolved? Can we close this?

AustinShalit, 2016-01-06:
The master branch is currently using idea number 2.

JLLeitschuh, 2016-01-06:
I like 2, it looks like @ThomasJClark took executive privilege and went and changed it in the network tables PR.
I vote for 2. Anyone else want to put in a vote?

JLLeitschuh, 2016-01-06:
@ThomasJClark Can you remove the pork from #239

ThomasJClark, 2016-01-06:
?

JLLeitschuh, 2016-01-06:
You made the executive decision to resolve this issue the way you wanted to without us coming to a consensus.

closed 2016-01-06 by ThomasJClark