WPIRoboticsProjects/GRIP #264: RARE: Deserialization Exception

JLLeitschuh, 2016-01-04:
What were the actions performed prior to this error appearing?

Loaded this file:
```xml
<grip:Pipeline>
  <sources>
    <grip:Camera>
      <property name="deviceNumber" value="0"/>
    </grip:Camera>
  </sources>
  <steps>
    <grip:Step name="resize">
      <grip:Input step="0" socket="0"/>
      <grip:Input step="0" socket="1"/>
      <grip:Input step="0" socket="2">
        <value>0.3</value>
      </grip:Input>
      <grip:Input step="0" socket="3">
        <value>0.3</value>
      </grip:Input>
      <grip:Input step="0" socket="4">
        <value>INTER_LINEAR</value>
      </grip:Input>
      <grip:Output step="0" socket="0" previewed="true"/>
    </grip:Step>
    <grip:Step name="Desaturate">
      <grip:Input step="1" socket="0"/>
      <grip:Output step="1" socket="0" previewed="true"/>
    </grip:Step>
    <grip:Step name="Find Lines">
      <grip:Input step="2" socket="0"/>
      <grip:Output step="2" socket="0" previewed="true"/>
    </grip:Step>
  </steps>
  <connections>
    <grip:Connection>
      <grip:Output step="0" socket="0" previewed="true"/>
      <grip:Input step="1" socket="0"/>
    </grip:Connection>
    <grip:Connection>
      <grip:Output step="1" socket="0" previewed="true"/>
      <grip:Input step="2" socket="0"/>
    </grip:Connection>
    <grip:Connection>
      <grip:Output source="0" socket="0" previewed="false"/>
      <grip:Input step="0" socket="0"/>
    </grip:Connection>
  </connections>
</grip:Pipeline>
```

I've loaded this same file ~ 20 to 30 times and this is the first time that I experienced this problem.

Message: An event subscriber threw an exception

## System Info:

Property Name | Property 
 ----- | -----
GRIP Version | 0.3.0
javafx.version | 8.0.66
java.runtime.name | Java(TM) SE Runtime Environment
java.vm.version | 25.66-b17
java.vm.vendor | Oracle Corporation
java.vm.name | Java HotSpot(TM) 64-Bit Server VM
java.runtime.version | 1.8.0_66-b17
java.awt.graphicsenv | sun.awt.CGraphicsEnvironment
javafx.runtime.version | 8.0.66-b17
os.name | Mac OS X
os.version | 10.11.2
[... 75 lines cut ...]
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:58)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.EventUtil.fireEventImpl(EventUtil.java:74)
	at com.sun.javafx.event.EventUtil.fireEvent(EventUtil.java:54)
	at javafx.event.Event.fireEvent(Event.java:198)
	at javafx.scene.Scene$MouseHandler.process(Scene.java:3757)
	at javafx.scene.Scene$MouseHandler.access$1500(Scene.java:3485)
	at javafx.scene.Scene.impl_processMouseEvent(Scene.java:1762)
	at javafx.scene.Scene$ScenePeerListener.mouseEvent(Scene.java:2494)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler$MouseEventNotification.run(GlassViewEventHandler.java:352)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler$MouseEventNotification.run(GlassViewEventHandler.java:275)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler.lambda$handleMouseEvent$354(GlassViewEventHandler.java:388)
	at com.sun.javafx.tk.quantum.QuantumToolkit.runWithoutRenderLock(QuantumToolkit.java:389)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler.handleMouseEvent(GlassViewEventHandler.java:387)
	at com.sun.glass.ui.View.handleMouseEvent(View.java:555)
	at com.sun.glass.ui.View.notifyMouse(View.java:937)

```

AustinShalit, 2016-01-21:
@JLLeitschuh Have you seen this issue again?

JLLeitschuh, 2016-01-21:
No, this is really rare. We should still have a safety check around cvtcolor.
I consider the opencv raw functions as somewhat dangerous. They are prone to segfaulting if you give the right input.

SamCarlberg, 2017-02-10:
Is this still a problem? I've never encountered it.

JLLeitschuh, 2017-03-01:
Someone want to try loading this file a few dozen times?