WPIRoboticsProjects/GRIP #267: Headless Mode

codetheweb, 2016-01-05:
Does GRIP support being run in headless mode yet from the command line? I.E. can I do something like `./grip --headless field-config.grip`?  If not, is this being developed?

JLLeitschuh, 2016-01-05:
Currently, only somewhat.
We currently produce a JAR that can be run that is only the core library without the UI.
If you build from source you can find the file in `core/build/libs`

This will be added fully in v0.4.0 (hopefully released tomorrow).
In v0.4.0 you will be able to deploy to a remote device from within grip as well as stop and start the program.

JLLeitschuh, 2016-01-05:
@ThomasJClark Usually, I allow the person who opened the issue to close it. It's less dismissive and allows the person to feel like they've had the opportunity to ask everything they want to.
If it sits for more than a week without comment then you close it.

codetheweb, 2016-01-05:
@JLLeitschuh ok, thanks.

closed 2016-01-05 by codetheweb