WPIRoboticsProjects/GRIP #270: Appveyor tests keeps failing

JLLeitschuh, 2016-01-05:
This isn't the first time I've seen this problem.
```java
edu.wpi.grip.ui.components.ExceptionWitnessResponderButtonTest > testPopoverAppears FAILED 
    java.lang.RuntimeException: java.lang.NullPointerException
        at org.testfx.util.WaitForAsyncUtils.callConditionAndReturnResult(WaitForAsyncUtils.java:348)
        at org.testfx.util.WaitForAsyncUtils.waitFor(WaitForAsyncUtils.java:169)
        at edu.wpi.grip.ui.components.ExceptionWitnessResponderButtonTest.testPopoverAppears(ExceptionWitnessResponderButtonTest.java:56)
 
        Caused by:
        java.lang.NullPointerException
            at org.testfx.matcher.base.NodeMatchers.isVisible(NodeMatchers.java:123)
            at org.testfx.matcher.base.NodeMatchers.lambda$isVisible$30(NodeMatchers.java:64)
            at org.testfx.matcher.base.GeneralMatchers$1.matches(GeneralMatchers.java:47)
            at edu.wpi.grip.ui.components.ExceptionWitnessResponderButtonTest.lambda$testPopoverAppears$3(ExceptionWitnessResponderButtonTest.java:57)
            at org.testfx.util.WaitForAsyncUtils.callConditionAndReturnResult(WaitForAsyncUtils.java:345)
            ... 2 more
 
90 tests completed, 1 failed, 2 skipped 
```

AustinShalit, 2016-01-07:
Was this fixed by #300?

JLLeitschuh, 2016-01-08:
Nope.

JLLeitschuh, 2016-01-29:
This needs to be fixed in order for #450 to be complete.

JLLeitschuh, 2016-02-06:
New one.
```
edu.wpi.grip.ui.MainWindowTest > testShouldCreateNewOperationInPipelineView FAILED 
    java.util.concurrent.TimeoutException
 
    java.lang.NullPointerException
        at edu.wpi.grip.ui.MainWindowTest.tearDown(MainWindowTest.java:34)
```

AustinShalit, 2016-07-30:
@JLLeitschuh Any updates on this?

SamCarlberg, 2016-10-06:
I haven't seen any wrongly failing tests on appveyor for a while; I'm assuming it was an issue with the build environment. This can be closed unless @JLLeitschuh thinks there's something I missed

AustinShalit, 2016-10-06:
I agree.  I think this is solved via time

JLLeitschuh, 2016-10-06:
I think this is resolved. I haven't seen it fail in a long time.

closed 2016-10-06 by JLLeitschuh