WPIRoboticsProjects/GRIP #272: File with Image Source will not open

AustinShalit, 2016-01-05:
This exception is thrown when trying to open a file with an image as the source.  

Message: An event subscriber threw an exception

## System Info:

Property Name | Property 
 ----- | -----
GRIP Version | 0.3.0
javafx.version | 8.0.66
java.runtime.name | Java(TM) SE Runtime Environment
java.vm.version | 25.66-b17
java.vm.vendor | Oracle Corporation
java.vm.name | Java HotSpot(TM) 64-Bit Server VM
java.runtime.version | 1.8.0_66-b17
java.awt.graphicsenv | sun.awt.CGraphicsEnvironment
javafx.runtime.version | 8.0.66-b17
os.name | Mac OS X
os.version | 10.11.2
os.arch | x86_64
file.encoding | UTF-8
java.vm.info | mixed mode
java.version | 1.8.0_66
sun.arch.data.model | 64
sun.cpu.endian | little

## Stack Trace:
```java
java.lang.IllegalStateException: Failed to load FXML
	at edu.wpi.grip.ui.GRIPUIModule$1.lambda$hear$37(GRIPUIModule.java:48)
	at com.google.inject.internal.MembersInjectorImpl.notifyListeners(MembersInjectorImpl.java:119)
	at com.google.inject.internal.ConstructorInjector.provision(ConstructorInjector.java:115)
	at com.google.inject.internal.ConstructorInjector.construct(ConstructorInjector.java:85)
	at com.google.inject.internal.ConstructorBindingImpl$Factory.get(ConstructorBindingImpl.java:267)
	at com.google.inject.internal.InjectorImpl$2$1.call(InjectorImpl.java:1016)
	at com.google.inject.internal.InjectorImpl.callInContext(InjectorImpl.java:1092)
	at com.google.inject.internal.InjectorImpl$2.get(InjectorImpl.java:1012)
	at com.google.inject.assistedinject.FactoryProvider2.invoke(FactoryProvider2.java:770)
	at com.sun.proxy.$Proxy20.create(Unknown Source)
	at edu.wpi.grip.ui.pipeline.source.SourceControllerFactory.create(SourceControllerFactory.java:37)
	at edu.wpi.grip.ui.pipeline.PipelineController.lambda$onSourceAdded$96(PipelineController.java:231)
	at edu.wpi.grip.ui.util.GRIPPlatform.runAsSoonAsPossible(GRIPPlatform.java:64)
	at edu.wpi.grip.ui.pipeline.PipelineController.onSourceAdded(PipelineController.java:230)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:497)
	at com.google.common.eventbus.EventSubscriber.handleEvent(EventSubscriber.java:74)
	at com.google.common.eventbus.SynchronizedEventSubscriber.handleEvent(SynchronizedEventSubscriber.java:47)
	at com.google.common.eventbus.EventBus.dispatch(EventBus.java:322)
	at com.google.common.eventbus.EventBus.dispatchQueuedEvents(EventBus.java:304)
	at com.google.common.eventbus.EventBus.post(EventBus.java:275)
	at edu.wpi.grip.core.serialization.SourceConverter.unmarshal(SourceConverter.java:55)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:72)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:66)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:50)
	at com.thoughtworks.xstream.converters.collections.AbstractCollectionConverter.readItem(AbstractCollectionConverter.java:71)
	at com.thoughtworks.xstream.converters.collections.CollectionConverter.addCurrentElementToCollection(CollectionConverter.java:98)
	at com.thoughtworks.xstream.converters.collections.CollectionConverter.populateCollection(CollectionConverter.java:91)
	at com.thoughtworks.xstream.converters.collections.CollectionConverter.populateCollection(CollectionConverter.java:85)
	at com.thoughtworks.xstream.converters.collections.CollectionConverter.unmarshal(CollectionConverter.java:80)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:72)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:66)
	at com.thoughtworks.xstream.converters.reflection.AbstractReflectionConverter.unmarshallField(AbstractReflectionConverter.java:480)
	at com.thoughtworks.xstream.converters.reflection.AbstractReflectionConverter.doUnmarshal(AbstractReflectionConverter.java:412)
	at com.thoughtworks.xstream.converters.reflection.AbstractReflectionConverter.unmarshal(AbstractReflectionConverter.java:263)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:72)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:66)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:50)
	at com.thoughtworks.xstream.core.TreeUnmarshaller.start(TreeUnmarshaller.java:134)
[... 62 lines cut ...]
	at javafx.fxml.FXMLLoader.loadImpl(FXMLLoader.java:2441)
	at javafx.fxml.FXMLLoader.loadImpl(FXMLLoader.java:3214)
	at javafx.fxml.FXMLLoader.loadImpl(FXMLLoader.java:3175)
	at javafx.fxml.FXMLLoader.load(FXMLLoader.java:3165)
	at edu.wpi.grip.ui.GRIPUIModule$1.lambda$hear$37(GRIPUIModule.java:42)
	... 97 more
Caused by: java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:497)
	at sun.reflect.misc.Trampoline.invoke(MethodUtil.java:71)
	at sun.reflect.GeneratedMethodAccessor3.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:497)
	at sun.reflect.misc.MethodUtil.invoke(MethodUtil.java:275)
	at javafx.fxml.FXMLLoader.loadImpl(FXMLLoader.java:2566)
	... 102 more
Caused by: java.lang.IllegalStateException: The images were never loaded from the filesystem. Must call `loadImage` first.
	at edu.wpi.grip.core.sources.ImageFileSource.createOutputSockets(ImageFileSource.java:100)
	at edu.wpi.grip.core.Source.getOutputSockets(Source.java:62)
	at edu.wpi.grip.ui.pipeline.source.SourceController.initialize(SourceController.java:74)
	... 112 more

```

JLLeitschuh, 2016-01-05:
I bet this is some sort of racetime condition.

AustinShalit, 2016-01-05:
Solved with #275

PR #275 (2016-01-05): Fix file source not deserializing correctly in UI
Closes #272
commit 09318d55 (2016-01-05): Fix file source not deserializing correctly in UI

Closes #272
closed 2016-01-05 by JLLeitschuh