WPIRoboticsProjects/GRIP #273: Enlarged menu dropdown

AustinShalit, 2016-01-05:
[image]

What causes this error is dragging the window from a large monitor to a smaller monitor.  See image below for monitor setup.  

[image]

For reference I am dragging the window from the middle monitor to the left.  After some testing, this is only a bug when going from large to small.  Going from small to large looks normal.

JLLeitschuh, 2016-01-05:
It seems that our DPI check can't be a static field. But instead needs to be recalculated every time.

ThomasJClark, 2016-01-05:
That's funny, since `DPIUtility` should return constant values on OSX, since we only do manually HiDPI adjustments on Linux.  Maybe it's weirdness in how JavaFX handles HiDPI.

Austin, is one of your screens a Retina display?

AustinShalit, 2016-01-05:
Yes.  What I have been calling the smaller screen is a Retina display.

AustinShalit, 2016-01-07:
I am working on this

AustinShalit, 2016-01-08:
This appears to be a JavaFX issue: https://bugs.openjdk.java.net/browse/JDK-8087565

AustinShalit, 2016-01-08:
Another link but they came to have fixed the issue: https://bugs.openjdk.java.net/browse/JDK-8095036

JLLeitschuh, 2016-01-08:
This seems to be a bug in javafx and not something we can fix.

PaulaRudy, 2016-01-10:
Nice find @AustinShalit! Thanks again for your help!

SamCarlberg, 2016-10-31:
@AustinShalit can you test this and see if it's still happening?

AustinShalit, 2016-10-31:
It is

JLLeitschuh, 2016-10-31:
I think we need to make all of the DPI properties get re-computed when the monitor changes.
This means the values for DPI screens can't be set statically, they would have to be listenable properties that can be changed.

Don't know how much of a refactor this would require.

SamCarlberg, 2016-10-31:
It's a bug in JavaFX with rendering the menu items. It's not something we can fix in GRIP.

JLLeitschuh, 2016-10-31:
Can we trigger a re-draw when the user changes screens?

SamCarlberg, 2016-10-31:
I don't think that would fix it. It's a problem with JavaFX not correctly updating DPI info when changing screens AFAICT

AustinShalit, 2020-04-07:
Bug in dependency. Closing

closed 2020-04-07 by AustinShalit