WPIRoboticsProjects/GRIP #280: "http://" before camera address

AustinShalit, 2016-01-06:
Teams are required to type http:// before typing the camera ip address.  Do we expect them to open any other kind of stream?  Can we fill the "http://" for them?

AustinShalit, 2016-01-06:
I will fix this

ThomasJClark, 2016-01-06:
Sounds good.

JLLeitschuh, 2016-01-06:
@nightpool Is this true?

ThomasJClark, 2016-01-06:
Even if it's not, we just make it only fill in the `http://` if nothing else is specified, so we don't break compatibility with other things

nightpool, 2016-01-06:
@JLLeitschuh uh can't think of any reason why not, but i mean it depends on how many kinds of protocols the underlying connection can support.

wouldn't the more elegent solution be to fill in `http://` if and only if they don't specify one?

AustinShalit, 2016-01-06:
#284 Solves this

closed 2016-01-07 by JLLeitschuh

AustinShalit, 2016-01-07:
thanks for closing