WPIRoboticsProjects/GRIP #281: text previews don't work?

nightpool, 2016-01-06:
- Open a new GRIP Project
 - Add a webcam
 - select the eyeball next to framerate
 - expect to see the framerate
 - GRIP continually displays an error.

(meta point: because the error occurs continuously, the only possible action is to close GRIP and restart it :/ maybe worth logging a separate ticket about for usability concerns? I know steps have a different way of reporting "something went wrong" that doesn't spam the view, so maybe its worth doing the same for previews?)

----

Message: An event subscriber threw an exception

## System Info:

Property Name | Property 
 ----- | -----
GRIP Version | 0.3.0
javafx.version | 8.0.66
java.runtime.name | Java(TM) SE Runtime Environment
java.vm.version | 25.66-b17
java.vm.vendor | Oracle Corporation
java.vm.name | Java HotSpot(TM) 64-Bit Server VM
java.runtime.version | 1.8.0_66-b17
java.awt.graphicsenv | sun.awt.CGraphicsEnvironment
javafx.runtime.version | 8.0.66-b17
os.name | Mac OS X
os.version | 10.11.1
os.arch | x86_64
file.encoding | UTF-8
java.vm.info | mixed mode
java.version | 1.8.0_66
sun.arch.data.model | 64
sun.cpu.endian | little

## Stack Trace:
```java
java.lang.NullPointerException
	at edu.wpi.grip.ui.preview.TextAreaSocketPreviewView.onSocketChanged(TextAreaSocketPreviewView.java:37)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:497)
	at com.google.common.eventbus.EventSubscriber.handleEvent(EventSubscriber.java:74)
	at com.google.common.eventbus.SynchronizedEventSubscriber.handleEvent(SynchronizedEventSubscriber.java:47)
	at com.google.common.eventbus.EventBus.dispatch(EventBus.java:322)
	at com.google.common.eventbus.EventBus.dispatchQueuedEvents(EventBus.java:304)
	at com.google.common.eventbus.EventBus.post(EventBus.java:275)
	at edu.wpi.grip.core.Socket.setValueOptional(Socket.java:70)
	at edu.wpi.grip.core.OutputSocket.setValueOptional(OutputSocket.java:36)
	at edu.wpi.grip.core.Socket.setValue(Socket.java:79)
	at edu.wpi.grip.core.OutputSocket.setValue(OutputSocket.java:41)
	at edu.wpi.grip.core.sources.CameraSource.lambda$start$15(CameraSource.java:197)
	at java.lang.Thread.run(Thread.java:745)

```

JLLeitschuh, 2016-01-06:
Sorry, I'm going to close this issue. Its already been fixed. :p
Build from master.

JLLeitschuh, 2016-01-06:
Duplicate of #254

nightpool, 2016-01-06:
quick resolution :D

closed 2016-01-06 by nightpool