WPIRoboticsProjects/GRIP #283: resize types don't work

nightpool, 2016-01-06:
not sure if this is indicative of a lower-level opencv bug, but the bottom three values of the "resize" operation do nothing

(on master)

stack trace from the logs:

OpenCV Error: Bad argument (Unknown interpolation method) in resize, file /Users/saudet/projects/bytedeco/javacpp-presets/opencv/cppbuild/macosx-x86_64/opencv-3.0.0/modules/imgproc/src/imgwarp.cpp, line 3363
java.lang.RuntimeException: /Users/saudet/projects/bytedeco/javacpp-presets/opencv/cppbuild/macosx-x86_64/opencv-3.0.0/modules/imgproc/src/imgwarp.cpp:3363: error: (-5) Unknown interpolation method in function resize

	at org.bytedeco.javacpp.opencv_imgproc.resize(Native Method)
	at edu.wpi.grip.generated.opencv_imgproc.Resize.perform(Resize.java:59)
	at edu.wpi.grip.core.Operation.perform(Operation.java:61)
	at edu.wpi.grip.core.Step.runPerformIfPossible(Step.java:145)
	at edu.wpi.grip.core.Step.onInputSocketChanged(Step.java:164)
	at sun.reflect.GeneratedMethodAccessor30.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:497)
	at com.google.common.eventbus.EventSubscriber.handleEvent(EventSubscriber.java:74)
	at com.google.common.eventbus.SynchronizedEventSubscriber.handleEvent(SynchronizedEventSubscriber.java:47)
	at com.google.common.eventbus.EventBus.dispatch(EventBus.java:322)
	at com.google.common.eventbus.EventBus.dispatchQueuedEvents(EventBus.java:304)
	at com.google.common.eventbus.EventBus.post(EventBus.java:275)
	at edu.wpi.grip.core.Socket.setValueOptional(Socket.java:70)
	at edu.wpi.grip.core.OutputSocket.setValueOptional(OutputSocket.java:36)
	at edu.wpi.grip.core.Socket.setValue(Socket.java:79)
	at edu.wpi.grip.core.OutputSocket.setValue(OutputSocket.java:41)
	at edu.wpi.grip.core.sources.CameraSource.lambda$start$15(CameraSource.java:209)
	at edu.wpi.grip.core.sources.CameraSource$$Lambda$402/1745982941.run(Unknown Source)
	at java.lang.Thread.run(Thread.java:745)

Property Name | Property 
 ----- | -----
GRIP Version | 0.3.0
javafx.version | 8.0.45
java.runtime.name | Java(TM) SE Runtime Environment
java.vm.version | 25.45-b02
java.vm.vendor | Oracle Corporation
java.vm.name | Java HotSpot(TM) 64-Bit Server VM
java.runtime.version | 1.8.0_45-b14
java.awt.graphicsenv | sun.awt.CGraphicsEnvironment
javafx.runtime.version | 8.0.45-b11
os.name | Mac OS X
os.version | 10.11.1
os.arch | x86_64
file.encoding | UTF-8
java.vm.info | mixed mode
java.version | 1.8.0_45
sun.arch.data.model | 64
sun.cpu.endian | little

JLLeitschuh, 2016-01-06:
I've tested this before and it works. Huh. I'll have to try that out and see what happens.
Thanks for the issue.

AustinShalit, 2016-01-06:
@nightpool What arguments are you putting in the boxes?  Here is a working example:
[image]

nightpool, 2016-01-06:
edited issue with more explicit description re: @AustinShalit's confusion

nightpool, 2016-01-06:
also some border types in `erode` have this issue:

- border_wrap
- border_transparent
- border_isolated

ThomasJClark, 2016-01-06:
@JLLeitschuh I recommend removing the `borderType` param in erode (I doubt anyone is using it)

JLLeitschuh, 2016-01-06:
Not in this release. Perhaps in the next one.

ThomasJClark, 2016-01-06:
Ok, but fixing the resize types should definitely happen.  That's a pretty bad bug

ThomasJClark, 2016-01-06:
Really both of them are pretty critical.  They should definitely be fixed before 1.0.0

PaulaRudy, 2016-01-06:
See also issue #151

nightpool, 2016-01-06:
So basically, @JLLeitschuh opencv is just really bad at types? For some reason I'm not at all surprised...

ThomasJClark, 2016-01-06:
This isn't an OpenCV issue, the issue is that we're including invalid values in the dropdown menu

nightpool, 2016-01-06:
@ThomasJClark sure, but the root cause of that is that OpenCV does not have well-defined enum types. (For example, they have a enum ThresholdTypes, but that enum is reused in places where it's not appropriate, such as AdaptiveThreshold)

JLLeitschuh, 2016-01-06:
Yes, opencv is really bad with types. We are generating that list dynamically from the sourcecode of another library that unfortunately contains list of integers instead of enums that we can use.

YIFanH, 2017-01-09:
@JLLeitschuh Hi,the enums types can not be found, you removed the enums to integers,could you give me an examples ? thank you for your repay!

JLLeitschuh, 2017-01-09:
I'm not sure what you mean? What are you trying to do with GRIP Where the enums aren't being found?

YIFanH, 2017-01-10:
@JLLeitschuh I am so sorry for my expressions,i make opencv_improc modules as a independent module in my android-ndk project, when used the resize function, the unknown interpolation method appeared.