WPIRoboticsProjects/GRIP #290: Document how to launch GRIP from FRC user program

JLLeitschuh, 2016-01-07:
Someone needs to do this.

AustinShalit, 2016-01-07:
```
Runtime.getRuntime().exec(params);
```

JLLeitschuh, 2016-01-07:
This should be put in the wiki.

AustinShalit, 2016-01-09:
Java Example:
Process grip = Runtime.getRuntime().exec(new String[]{"/usr/local/frc/JRE/bin/java", "-jar", "grip.jar", FILENAME_AS_STRING});

ThomasJClark, 2016-01-09:
Thanks @AustinShalit, I'll put that it in the wiki before release

JLLeitschuh, 2016-01-09:
I already did.

PaulaRudy, 2016-01-10:
Can this be closed or does this need further attention?

JLLeitschuh, 2016-01-15:
This document may need work but it's there.
https://github.com/WPIRoboticsProjects/GRIP/wiki/Tutorial:-Run-GRIP-from-a-CPP-or-Java-FRC-program
I think that the java one still needs to be tested fully.
I think this one is a duplicate issue: #354
When it is resolved I think we can close this one.

AustinShalit, 2016-07-29:
I believe this has been solved.

closed 2016-07-29 by AustinShalit