WPIRoboticsProjects/GRIP #297: Process does not terminate when UI is closed

jkoehring, 2016-01-07:
I just installed GRIP-0.3.0.exe on Windows 7. When I terminate the program, either by selecting File => Exit or by clicking on the red X in the upper-right corner of the window, the UI closes but the process does not terminate. I am forced to terminate the GRIP.exe process using Windows Task Manager.

JLLeitschuh, 2016-01-07:
If you are using a webcam does your webcam light stay on?

jkoehring, 2016-01-07:
So far, I have not used a webcam with GRIP. I have just loaded image files from disk.

jkoehring, 2016-01-07:
I will also add that the orphaned process consumes about 13% of the CPU on a Lenovo W530 laptop with a 2.8 GHz quad-core i7 processor. So, it's not just sitting there doing nothing.

JLLeitschuh, 2016-01-07:
@PeterJohnson 
Could this be a network tables problem???

jkoehring, 2016-01-07:
FYI, I was running GRIP.exe standalone. There was no other FRC software running (e.g., Driver Station, Eclipse, ...) nor was there a roboRIO connected to my computer.

JLLeitschuh, 2016-01-07:
Ooooooohhhh... v0.3.0 was prior to us adding network tables so that can't be related.
@PeterJohnson You're off the hook.

333fred, 2016-01-07:
These are some profiles from the application after the main window has been closed on Windows 10. My process was open the application, connect VisualVM (https://visualvm.java.net/), and close the application. I then waited a while and did an export of the application. To enable profiling in Windows, you need to add `-Dcom.sun.management.jmxremote` to the JVMOptions section (new line) in GRIP.cfg in %APP_DATA%\local\GRIP\app. Other platforms will need similar, not sure where those locations will be.
[grip_profiles.zip](https://github.com/WPIRoboticsProjects/GRIP/files/81719/grip_profiles.zip)

PaulaRudy, 2016-01-07:
I'm encountering this as well on Ubuntu- I will close the window but GRIP keeps spamming the console with 
`Jan 07, 2016 4:52:41 PM edu.wpi.grip.core.operations.networktables.NTManager lambda$new$9`
`SEVERE: ntcore/src/tcpsockets/TCPConnector.cpp:157 select() to roboio-0-frc.local port 1735 error 111 - Connection refused`

and occasionally

`Jan 07, 2016 4:52:41 PM edu.wpi.grip.ui.util.GRIPPlatform onJavaFXRunnerEvent`
`WARNING: POTENTIAL DEADLOCK!`

JLLeitschuh, 2016-01-08:
@jkoehring If I publish another release tonight or tomorrow morning will you have the opportunity to test this out again?

jkoehring, 2016-01-08:
I would be happy to test the latest build, but there doesn't seem to be a Windows installer available.

JLLeitschuh, 2016-01-08:
I just created another release. We were having problems with our windows build system.
This should be fixed in ~ 12 min. (v1.0.0-rc4)[https://github.com/WPIRoboticsProjects/GRIP/releases/tag/v1.0.0-rc4]

jkoehring, 2016-01-08:
I just installed 1.0.0-rc4. File=>Exit (and its shortcut Ctrl+Q) now work, but clicking on the Close (X) in the upper right-hand corner of the window now hangs the application and changes its status to Not Responding. Clicking the Close (X) again causes Windows to display the "not responding" dialog, letting you to force close the application.

333fred, 2016-01-08:
It's stuck on the `PlatformImpl.runAndWait()` call in Main.java, line 86.

333fred, 2016-01-08:
Full Thread Dump:
[grip_thread_dump.txt](https://github.com/WPIRoboticsProjects/GRIP/files/83128/grip_thread_dump.txt)

JLLeitschuh, 2016-01-09:
The exception that the program is responding to is an IO Exception.

```java
java.lang.RuntimeException: java.io.IOException: Cannot run program "$HOME\AppData\Local\GRIP\runtime/bin/java": CreateProcess error=2, The system cannot find the file specified
	at org.bytedeco.javacpp.Loader$2.run(Loader.java:645)
Caused by: java.io.IOException: Cannot run program "$HOME\AppData\Local\GRIP\runtime/bin/java": CreateProcess error=2, The system cannot find the file specified
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at org.bytedeco.javacpp.Loader$2.run(Loader.java:643)
Caused by: java.io.IOException: CreateProcess error=2, The system cannot find the file specified
	at java.lang.ProcessImpl.create(Native Method)
	at java.lang.ProcessImpl.<init>(ProcessImpl.java:386)
	at java.lang.ProcessImpl.start(ProcessImpl.java:137)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 1 more
```

JLLeitschuh, 2016-01-09:
Opened an issue with bytedeco: https://github.com/bytedeco/javacpp/issues/60

PR #308 (2016-01-08): Fix windows not shutting down when application stopped
Closes #297
commit 08e43ed4 (2016-01-08): Fix windows not shutting down when application stopped

Closes #297
PR #334 (2016-01-09): Fix windows hanging when being shutdown
This was caused because a shtdown hook threw an exception.
Our main application is designed to try to handle threads dying by
displaying a UI element. This should not be done in a shutdown hook.
What we have to do is ensure that when we try dto display the
exception alert it isn't from inside a shutdown hook.

Closes #297
commit 694d9e42 (2016-01-09): Fix windows hanging when being shutdown

This was caused because a shtdown hook threw an exception.
Our main application is designed to try to handle threads dying by
displaying a UI element. This should not be done in a shutdown hook.
What we have to do is ensure that when we try dto display the
exception alert it isn't from inside a shutdown hook.

Closes #297
closed 2016-01-09 by ThomasJClark