WPIRoboticsProjects/GRIP #309: TOO MANY LOG FILES!

JLLeitschuh, 2016-01-08:
I have over 50 Grip.log files in my home directory.
We need to implement some sort of versioning so that this doesn't happen.

PaulaRudy, 2016-01-09:
I'm not having this issue on Ubuntu. You're on apple, right?

JLLeitschuh, 2016-01-09:
[image]

Yes

JLLeitschuh, 2016-01-09:
I have 50 of these files.

AustinShalit, 2016-01-09:
I can confirm this is happening to me too.  I am on OS X as well.

PaulaRudy, 2016-01-10:
Apparently these results are cumulative (info courtesy of @ThomasJClark).

PaulaRudy, 2016-01-10:
According to the documentation and the latest code I pulled,  tell this should never be happening. @JLLeitschuh Is GRIPCoreModule() constructed more than once? That would be the only way I could see this happening.

JLLeitschuh, 2016-01-10:
I don't think so. I think we would see the the "Config complete" more than once.

PaulaRudy, 2016-01-11:
Looks like this is triggered by the tests somehow.... running ./gradlew check will trigger this issue for me.

PR #360 (2016-01-15): Fixes tests creating multiple log files in home directory
The problem was we were setting up the logger multiple times
durring the test's lifecycle.
We should only have one global logger setup durring the tests.

Closes #309
commit 320776db (2016-01-15): Fixes tests creating multiple log files in home directory

The problem was we were setting up the logger multiple times
durring the test's lifecycle.
We should only have one global logger setup durring the tests.

Closes #309
closed 2016-01-15 by JLLeitschuh