WPIRoboticsProjects/GRIP #316: Create complex pipeline file to test against compatibility breaking

JLLeitschuh, 2016-01-08:
We need to create a grip save file with almost everything inside of it (except for perhaps camera sources) that we can add to the test system. This file will ensure that we don't break compatibility during the season.

PaulaRudy, 2016-01-10:
I can take a stab at this.

PaulaRudy, 2016-01-15:
Did we need all the operations chained together? Or will just giving them enough appropriate input to execute be enough?

JLLeitschuh, 2016-01-15:
Uhhh. I would connect a few of them, sure. You don't need to connect all of them though no.

AustinShalit, 2016-01-20:
I am going to hop on this

PaulaRudy, 2016-01-23:
So! Here is the file:

```
<grip:Pipeline>
  <sources>
    <grip:ImageFile>
      <property name="path" value="REPLACEME"/>
    </grip:ImageFile>
    <grip:Camera>
      <property name="deviceNumber" value="0"/>
    </grip:Camera>
  </sources>
  <steps>
    <grip:Step name="Resize Image">
      <grip:Input step="0" socket="0"/>
      <grip:Input step="0" socket="1">
        <value>200.0</value>
      </grip:Input>
      <grip:Input step="0" socket="2">
        <value>100.0</value>
      </grip:Input>
      <grip:Input step="0" socket="3">
        <value>CUBIC</value>
      </grip:Input>
      <grip:Output step="0" socket="0" previewed="true"/>
    </grip:Step>
    <grip:Step name="Blur">
      <grip:Input step="1" socket="0"/>
      <grip:Input step="1" socket="1">
        <value>BOX</value>
      </grip:Input>
      <grip:Input step="1" socket="2">
        <value>46.846846846846844</value>
      </grip:Input>
      <grip:Output step="1" socket="0" previewed="true"/>
    </grip:Step>
    <grip:Step name="Desaturate">
      <grip:Input step="2" socket="0"/>
      <grip:Output step="2" socket="0" previewed="true"/>
    </grip:Step>
    <grip:Step name="RGB Threshold">
      <grip:Input step="3" socket="0"/>
      <grip:Input step="3" socket="1">
        <value>
          <double>82.55395683453237</double>
          <double>255.0</double>
        </value>
      </grip:Input>
      <grip:Input step="3" socket="2">
        <value>
          <double>41.0402645622285</double>
          <double>157.3718868704939</double>
        </value>
      </grip:Input>
      <grip:Input step="3" socket="3">
        <value>
          <double>2.039137318027521</double>
          <double>130.05461288324597</double>
        </value>
      </grip:Input>
      <grip:Output step="3" socket="0" previewed="true"/>
    </grip:Step>
    <grip:Step name="HSV Threshold">
      <grip:Input step="4" socket="0"/>
      <grip:Input step="4" socket="1">
        <value>
          <double>2.4242914789903782</double>
          <double>101.48903968042924</double>
        </value>
      </grip:Input>
[... 763 lines cut ...]
    </grip:Connection>
    <grip:Connection>
      <grip:Output step="0" socket="0" previewed="true"/>
      <grip:Input step="13" socket="0"/>
    </grip:Connection>
  </connections>
  <settings>
    <teamNumber>0</teamNumber>
    <publishAddress>roborio-0-frc.local</publishAddress>
    <deployAddress>roborio-0-frc.local</deployAddress>
    <deployDir>/home/lvuser</deployDir>
    <deployUser>lvuser</deployUser>
    <deployJavaHome>/usr/local/frc/JRE/</deployJavaHome>
  </settings>
</grip:Pipeline>
```

This file requires the following image:
[image]

Note that you have to replace the REPLACEME text in the file with the absolute path to the image.

This file also requires a camera at 0. 

Note that the config stuff at the bottom may need to be altered for OSes other than linux.

PaulaRudy, 2016-01-23:
Unfortunately, because of the size of the file, it throws errors when you load it (the steps think there is no input because it hasn't initialized yet. So I'm not sure how to use this in a test? Maybe wait and then check? Any thoughts?

closed 2016-01-28 by ThomasJClark