WPIRoboticsProjects/GRIP #325: Mouse Follow "Move Step"

cpapplefamily, 2016-01-09:
I just started to test the tool you created. I have been trying the different Palette tools. I could see a real use if the mouse pointer stayed over the Move Left/Right with the selected tool.

ThomasJClark, 2016-01-09:
That's a good point, the current way you move steps is kind of cumbersome.  I think in the long term what we want to do is let you actually drag and drop steps.

cpapplefamily, 2016-01-09:
Even better would be drag and drop. I been struggling to test the OpenCV functions on the Raspberry Pi for the past few weeks reading myself for this season. Love the live real time display.

SamCarlberg, 2016-12-08:
Drag and drop was added a while back and didn't close this issue

closed 2016-12-08 by SamCarlberg