WPIRoboticsProjects/GRIP #327: Another Useful option "Minimize Filter"

cpapplefamily, 2016-01-09:
If each operation had an option to only display the used Input and Output connectors with a limited label the box could be reduced to a fraction of the size saving screen restate. 

? Is this the right place for these ideas?

JLLeitschuh, 2016-01-09:
Yes, this is the right place for the ideas. If you want to discuss how to actually implement these ideas then join us in the [gitter chat](https://gitter.im/WPIRoboticsProjects/GRIP)

Could you provide a sketch for clarity?

cpapplefamily, 2016-01-09:
Full Sized
[image]

cpapplefamily, 2016-01-09:
Minimized
[image]

cpapplefamily, 2016-01-09:
Could Stack them too then.

JLLeitschuh, 2016-01-09:
Nice photoshop! This makes a lot more sense now!!!

PR #659 (2016-08-11): Osulacam minimize steps
closes #327 
closes #639 

Fix-up of #639
closed 2016-08-11 by AustinShalit