WPIRoboticsProjects/GRIP #333: Gradle Build in Eclipse Spams console with the following error message

rlee287, 2016-01-09:
`java.lang.StringIndexOutOfBoundsException: String index out of range: -1
	at java.lang.String.substring(String.java:1927)
	at edu.wpi.grip.core.operations.networktables.NTManager.lambda$new$8(NTManager.java:46)`

ThomasJClark, 2016-01-09:
Did you do anything in the application before the error started showing up?

JLLeitschuh, 2016-01-09:
We use Intellij for all development we would recommend using that.

JLLeitschuh, 2016-01-09:
I'm getting the same problem on master when I start up grip from the command line. I'm running windows.

rlee287, 2016-01-09:
It starts spamming right as the GRIP window opens. I am building this with the Eclipse Gradle Plugins. It occasionally throws a "NTLogger" thing after quitting GRIP. This is the complete build log:
[sts] -----------------------------------------------------
[sts] Starting Gradle build for the following tasks: 
[sts]      :ui:run
[sts] -----------------------------------------------------
:buildSrc:compileJava UP-TO-DATE
:buildSrc:extractSources UP-TO-DATE
:buildSrc:compileGroovy UP-TO-DATE
:buildSrc:processResources UP-TO-DATE
:buildSrc:classes UP-TO-DATE
:buildSrc:jar UP-TO-DATE
:buildSrc:assemble UP-TO-DATE
:buildSrc:compileTestJava UP-TO-DATE
:buildSrc:compileTestGroovy UP-TO-DATE
:buildSrc:processTestResources UP-TO-DATE
:buildSrc:testClasses UP-TO-DATE
:buildSrc:test UP-TO-DATE
:buildSrc:check UP-TO-DATE
:buildSrc:build UP-TO-DATE
:core:generateCodeFromSource
:core:compileJava UP-TO-DATE
:core:processResources UP-TO-DATE
:core:classes UP-TO-DATE
:core:shadowJar UP-TO-DATE
:ui:compileJava UP-TO-DATE
:ui:processResources UP-TO-DATE
:ui:cssToBin SKIPPED
:ui:classes UP-TO-DATE
:ui:run
Jan 09, 2016 12:03:12 PM java.util.logging.LogManager$RootLogger log
CONFIG: Configuration done.
platform: /Windows/amd64/
java.lang.StringIndexOutOfBoundsException: String index out of range: -1
	at java.lang.String.substring(String.java:1927)
	at edu.wpi.grip.core.operations.networktables.NTManager.lambda$new$8(NTManager.java:46)
java.lang.StringIndexOutOfBoundsException: String index out of range: -1
	at java.lang.String.substring(String.java:1927)
	at edu.wpi.grip.core.operations.networktables.NTManager.lambda$new$8(NTManager.java:46)
java.lang.StringIndexOutOfBoundsException: String index out of range: -1
	at java.lang.String.substring(String.java:1927)
	at edu.wpi.grip.core.operations.networktables.NTManager.lambda$new$8(NTManager.java:46)
java.lang.StringIndexOutOfBoundsException: String index out of range: -1
	at java.lang.String.substring(String.java:1927)
	at edu.wpi.grip.core.operations.networktables.NTManager.lambda$new$8(NTManager.java:46)
java.lang.StringIndexOutOfBoundsException: String index out of range: -1
	at java.lang.String.substring(String.java:1927)
	at edu.wpi.grip.core.operations.networktables.NTManager.lambda$new$8(NTManager.java:46)
java.lang.StringIndexOutOfBoundsException: String index out of range: -1
	at java.lang.String.substring(String.java:1927)
	at edu.wpi.grip.core.operations.networktables.NTManager.lambda$new$8(NTManager.java:46)
java.lang.StringIndexOutOfBoundsException: String index out of range: -1
	at java.lang.String.substring(String.java:1927)
	at edu.wpi.grip.core.operations.networktables.NTManager.lambda$new$8(NTManager.java:46)
java.lang.StringIndexOutOfBoundsException: String index out of range: -1
	at java.lang.String.substring(String.java:1927)
	at edu.wpi.grip.core.operations.networktables.NTManager.lambda$new$8(NTManager.java:46)
Exception in thread "NTLogger" Exception in thread "NTLogger" Exception in thread "NTLogger" Exception in thread "NTLogger" Exception in thread "NTLogger" Exception in thread "NTLogger" Exception in thread "NTLogger" Exception in thread "NTLogger" 
BUILD SUCCESSFUL

Total time: 18.961 secs
[sts] -----------------------------------------------------
[sts] Build finished succesfully!
[sts] Time taken: 0 min, 19 sec
[sts] -----------------------------------------------------

ThomasJClark, 2016-01-10:
Ah, I know the problem.

Our custom NetworkTables logger callback using a string search to find the filename that the error came from.  Native Windows code uses backwards slashes for paths, so this fails.

JLLeitschuh, 2016-01-10:
Awesome!

closed 2016-01-10 by JLLeitschuh