WPIRoboticsProjects/GRIP #344: Getting Started Questions

tpagithub, 2016-01-10:
The purpose of this question is to understand how vision processing can be used during a match. 

Can you give a brief explanation of how GRIP will be used to capture and process an image (e.g. reflective tape), then process that image and then what subsystem uses that image data and for what purposes?

bradamiller, 2016-01-10:
GRIP running on your robot or driver station computer connects to the camera on your robot. It has a pipeline that looks for targets or other objects of interest and generates "answers" (usually a set of contours or other results) that are published in Network Tables. Then your robot program can get that data and use it for aiming. Usually that last step, that is figuring out which goal is relevant, is done in code on your robot. You can see an example here: http://wpilib.screenstepslive.com/s/4485/m/50711/l/481750-using-grip-for-the-2016-game.
The sample images provided here:
https://usfirst.collab.net/sf/frs/do/listReleases/projects.wpilib/frs.2016_field_images
can be used for developing the GRIP program, but you should eventually modify the threshold and other values yourself for your camera and your LEDs.

tpagithub, 2016-01-11:
Thanks for the reply. I understand the network table part. That is data about the shape you are looking for and once found an action can be taken. 
What is a pipeline? It sounds like GRIP specific term. 
How should I use those pictures? Are they like seed data that I process using the GRIP Gui into network tables. I am not quite understanding the how my real time camera is supposed to find that particular U shape and then react. It seems to me that I process all those pictures into numeric values and they are just stored in files on the rio, and then during real time operation my camera's data is compared against the value of the network tables. Once a match is found between realtime data and the network tables I created at the beginning of the season using your pictures, I can use logic to pass some value  to my robot code thread which might make the robot do some action like stop or fire or both. Can you please elaborate? Thank you.

ThomasJClark, 2016-01-11:
> What is a pipeline? It sounds like GRIP specific term. 

"Pipeline" is just the term we use for a series of steps in grip.  Basically a grip project.

> How should I use those pictures? Are they like seed data that I process using the GRIP Gui into network tables. I am not quite understanding the how my real time camera is supposed to find that particular U shape and then react.

You can use sample image at first to test the pipeline on your computer, but you'd probably replace them with cameras eventually.  You would almost certainly re-connect any steps connected to images to cameras before deploying to a roborio.

It's probably a good idea to leave the image sources in the project even after you add camera sources though, since you can just reconnect them to test out changes later on.

tpagithub, 2016-01-11:
Thank you Thomas. Is the purpose of the gui to test robot code? I am having trouble understanding the real time relationship between The GRIP gui and the real time operation of my camera. (Keep in mind we've never done this before. So I have no experience.) I can't see my drive team during a match taking a picture and then manually generating a pipeline. Can you describe a real time usage? 

It looks like there is an interface in the GUI to the rio in the menu. Can you please describe how the gui interacts with the rio?  From what I have researched the gui pushes two things, image data in the form of a network table, and it writes the openCV libraries to a dir on the rio. That dir location would be referenced in the robot.java to access the network tables. Could you please confirm my understanding?

ThomasJClark, 2016-01-13:
> Thank you Thomas. Is the purpose of the gui to test robot code?

Basically yes.  You use the GUI to design and test the algorithm, then typically run it on the robot in headless mode.

> It looks like there is an interface in the GUI to the rio in the menu. Can you please describe how the gui interacts with the rio? From what I have researched the gui pushes two things, image data [...]

The deploy button in the GUI uploads a headless version of GRIP and your project file to the RIO.  Assuming you have a "Publish ..." operation in your project, it will publish values to NetworkTables that you can get in your user program.

closed 2016-01-15 by ThomasJClark