WPIRoboticsProjects/GRIP #345: Installing on Ubuntu 14.04 (in parallels)

bradamiller, 2016-01-10:
When trying to run GRIP after an Ubuntu install it doesn't work (nothing happens). Then if I go to /opt/GRIP where it installed and run GRIP I get this:

parallels@ubuntu:/opt/GRIP$ ./GRIP
Jan 10, 2016 1:04:04 PM java.util.logging.LogManager$RootLogger log
CONFIG: Configuration done.
platform: /Linux/amd64/
Exception in Application start method
GRIP Error invoking method.
GRIP Failed to launch JVM

parallels@ubuntu:/opt/GRIP$ java -version
openjdk version "1.8.0_72-internal"
OpenJDK Runtime Environment (build 1.8.0_72-internal-b05)
OpenJDK 64-Bit Server VM (build 25.72-b05, mixed mode)
parallels@ubuntu:/opt/GRIP$ ^C
parallels@ubuntu:/opt/GRIP$

ThomasJClark, 2016-01-10:
I'll try the same fix for Fedora (manually building ntcore)

PaulaRudy, 2016-01-10:
I'm getting:

```arsenic@Angel:/opt/GRIP$ ./GRIP 
Jan 10, 2016 4:42:52 PM java.util.logging.LogManager$RootLogger log
CONFIG: Configuration done.
platform: /Linux/amd64/
terminate called after throwing an instance of 'std::system_error'
  what():  Enable multithreading to use std::thread: Operation not permitted
Aborted (core dumped)
```

ThomasJClark, 2016-01-11:
This should be fixed now, try it out and let me know

https://github.com/WPIRoboticsProjects/GRIP/releases/tag/v1.0.0

PaulaRudy, 2016-01-11:
I'm getting this:
```
Exception in Application start method
GRIP Error invoking method.
GRIP Failed to launch JVM
```

Where are all the files installed anyway? They're all in opt, right?

FlamingSpork, 2016-01-14:
I'm getting the following on Ubuntu 12.04:
```
$ ./GRIP 
Jan 14, 2016 3:50:28 PM java.util.logging.LogManager$RootLogger log
CONFIG: Configuration done.
platform: /Linux/amd64/
Exception in Application start method
GRIP Error invoking method.
GRIP Failed to launch JVM
```
I'm using the following Java version:
```
$ java -version
java version "1.8.0_66"
Java(TM) SE Runtime Environment (build 1.8.0_66-b17)
Java HotSpot(TM) 64-Bit Server VM (build 25.66-b17, mixed mode)
```
Is it possible to get it to run under 12.04?

rich-coe, 2016-01-17:
I'm facing a similar issue installing on my linux version.
I ran ldd against one of the libNetworkTablesJNI*.so files extracted by GRIP when trying to load and 
I got the following messages:
ldd /tmp/libNetworkTablesJNI6252791166581577015.so 
/tmp/libNetworkTablesJNI6252791166581577015.so: /usr/lib64/libstdc++.so.6: version `GLIBCXX_3.4.20' not found (required by /tmp/libNetworkTablesJNI6252791166581577015.so)
/tmp/libNetworkTablesJNI6252791166581577015.so: /usr/lib64/libstdc++.so.6: version `CXXABI_1.3.8' not found (required by /tmp/libNetworkTablesJNI6252791166581577015.so)
/tmp/libNetworkTablesJNI6252791166581577015.so: /usr/lib64/libstdc++.so.6: version `CXXABI_1.3.9' not found (required by /tmp/libNetworkTablesJNI6252791166581577015.so)
/tmp/libNetworkTablesJNI6252791166581577015.so: /usr/lib64/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by /tmp/libNetworkTablesJNI6252791166581577015.so)
        linux-vdso.so.1 (0x00007fff7d9fd000)
        libstdc++.so.6 => /usr/lib64/libstdc++.so.6 (0x00007f6a9bc40000)
        libm.so.6 => /lib64/libm.so.6 (0x00007f6a9b93f000)
        libgcc_s.so.1 => /lib64/libgcc_s.so.1 (0x00007f6a9b727000)
        libc.so.6 => /lib64/libc.so.6 (0x00007f6a9b380000)
        /lib64/ld-linux-x86-64.so.2 (0x000056397cf5f000)

It would appear that there is a dependency on a specific version of libstdc++.

ThomasJClark, 2016-01-17:
> I'm facing a similar issue installing on my linux version.

What Linux version is this?

rich-coe, 2016-01-17:
@ThomasJClark  opensuse-13.2 
I realize that you are only supporting specific ubuntu versions, I only commented because I saw other complaints with a similar error.

ThomasJClark, 2016-01-17:
Ok.  I know that on Travis (which is Ubuntu 12.04) we install gcc-4.8 and g++-4.8 before each build.  I think it has something to do with the libc and libstdc++ version on old versions of ubuntu.

rich-coe, 2016-01-17:
I haven't pulled the gcc source versions for 4.8.4 or 4.8.5, but one of these versions must update the GLIBCXX level from 3.4.19 to 3.4.20 and the CXXABI level from 1.3.7 to 1.3.9, possibly version 4.8.5.
Note, gcc-4.9.0 has the same glibcxx/abi level as gcc-4.8.3.
I suspect, but have not proven, that the required libstdc++ version comes from gcc-4.8.5 / gcc-4.9.3.

multiplemonomials, 2016-01-20:
Same on Linux Mint 17.1

multiplemonomials, 2016-01-20:
Also, the libstdc++ ABI versions are listed at https://gcc.gnu.org/onlinedocs/libstdc++/manual/abi.html

cinzy, 2016-01-23:
I am also getting the following in Ubuntu 14.04:
Jan 23, 2016 3:16:35 PM java.util.logging.LogManager$RootLogger log
CONFIG: Configuration done.
Jan 23, 2016 3:16:35 PM java.util.logging.LogManager$RootLogger log
CONFIG: GRIP Version: 1.1.1
platform: /Linux/amd64/
Exception in Application start method
GRIP Error invoking method.
GRIP Failed to launch JVM

My java version:
java version "1.8.0_66"
Java(TM) SE Runtime Environment (build 1.8.0_66-b17)
Java HotSpot(TM) 64-Bit Server VM (build 25.66-b17, mixed mode)

g++ version:
g++ (Ubuntu 4.8.4-2ubuntu1~14.04) 4.8.4

gcc version:
gcc (Ubuntu 4.8.4-2ubuntu1~14.04) 4.8.4

multiplemonomials, 2016-01-24:
I rebuilt networktables on my Linux laptop and substituted in the jar file, but it still wouldn't load.  I got something like 
```
terminated, std::exception   what(): enable multithreading to use std::thread, operation not permitted
```

Then, I tried running the jar file directly:
```
java -jar ui-1.0.0.jar
```
It started up and worked fine.  So it looks like the problem is the Java FX launcher.  It looks like it, or something it calls into, isn't compiled with `-pthread`

ThomasJClark, 2016-01-24:
> It started up and worked fine. So it looks like the problem is the Java FX launcher. It, or something it calls into, isn't compiled with `-pthread`.

That's correct - NetworkTables isn't compiled with -pthread, which may or may not be a problem depending on the compiler, the system libraries, and the executable it's linked to in.  The version we ship in the GRIP installer up to this point has been manually built, but [this was fixed upstream recenterly](https://usfirst.collab.net/gerrit/#/c/1327/)

>     java -jar ui-1.0.0.jar

[FYI a newer release of GRIP is out](https://github.com/WPIRoboticsProjects/GRIP/releases)

PaulaRudy, 2016-01-25:
Would you like me to test this again?

PR #648 (2016-08-02): Add Launcher Binary
closes #566 
closes #345 
closes #370 

This solves the Linux build issues!

Thank you @PeterJohnson and @FibreFoX for your work on this.
closed 2016-08-02 by JLLeitschuh