WPIRoboticsProjects/GRIP #346: Unable to run GRIP on Ubuntu

docprofsky, 2016-01-11:
I am unable to run GRIP on Ubuntu 14.04.3 LTS with Java 1.8.0_66. I installed GRIP from the deb package for the 1.0.0 release.

If I run the GRIP executable in `/opt/GRIP` I get this error.
```
Jan 10, 2016 1:46:53 PM java.util.logging.LogManager$RootLogger log
CONFIG: Configuration done.
platform: /Linux/amd64/
Exception in Application start method
GRIP Error invoking method.
GRIP Failed to launch JVM
```

If I run the GRIP jar file at `app/ui-1.0.0.jar` I get this error.
https://gist.github.com/docprofsky/f4eaab601183b7d10333#file-grip-1-0-0-release-ubuntu-txt

I get the same error when using gradle.
https://gist.github.com/docprofsky/f4eaab601183b7d10333#file-grip-dev-ubuntu-error-txt

ThomasJClark, 2016-01-11:
This should be fixed now, try it out and let me know

https://github.com/WPIRoboticsProjects/GRIP/releases/tag/v1.0.0

davidtsong, 2016-08-20:
Still broken ^^

AustinShalit, 2016-08-21:
@davidtsong Would you mind testing the latest build? (I will post a link tomorrow morning)

AustinShalit, 2016-08-21:
@davidtsong Please try this download: https://www.dropbox.com/s/9p76qiqd88gh2kp/grip-1.4.0-x64.deb?dl=1

smile-on, 2016-12-04:
I have tried latest grip-1.5.0-x64.deb, still broken.
`$ /opt/GRIP/GRIP
Dec 03, 2016 8:36:11 PM java.util.logging.LogManager$RootLogger log
CONFIG: Configuration done.
Dec 03, 2016 8:36:11 PM java.util.logging.LogManager$RootLogger log
CONFIG: GRIP Version: 1.5.0
SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
platform: /Linux/amd64/
Exception in Application init method
GRIP Error invoking method.
GRIP Failed to launch JVM
`

Same with running jar file
`$ java -version
java version "1.8.0_112"
Java(TM) SE Runtime Environment (build 1.8.0_112-b15)
Java HotSpot(TM) 64-Bit Server VM (build 25.112-b15, mixed mode)
$ java -jar /opt/GRIP/app/GRIP-1.5.0-x64.jar
Dec 03, 2016 8:32:32 PM java.util.logging.LogManager$RootLogger log
CONFIG: Configuration done.
Dec 03, 2016 8:32:32 PM java.util.logging.LogManager$RootLogger log
CONFIG: GRIP Version: 1.5.0
SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
platform: /Linux/amd64/
Exception in Application init method
java.lang.reflect.InvocationTargetException
at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
at java.lang.Thread.run(Thread.java:745)
Caused by: com.google.inject.ProvisionException: Unable to provision, see the following errors:
Error injecting constructor, java.lang.UnsatisfiedLinkError: no ntcore in java.library.path
  at edu.wpi.grip.core.operations.network.networktables.NTManager.<init>(Unknown Source)
`
Full log is attached. 

>     at edu.wpi.first.wpilibj.networktables.NetworkTablesJNI.<clinit>(NetworkTablesJNI.java:57)
>     java.lang.UnsatisfiedLinkError: no **ntcore in java.library.path**

This error message points on some dependency **ntcore in java.library.path** that  was not installed. Could you clarify what additional software is expected to have before running GRIP on Ubuntu?

[grip-error.log.zip](https://github.com/WPIRoboticsProjects/GRIP/files/629233/grip-error.log.zip)