WPIRoboticsProjects/GRIP #348: Unable to Execute file GRIP.exe

jojoguy10, 2016-01-13:
One of laptops is running Windows 10 Pro 32-bit and it is coming up with this error when trying to install:

[image]

It installs fine on our other laptop that has Windows 10 Pro 64-bit

ThomasJClark, 2016-01-13:
GRIP is built for 64 bit windows right now.

237 ran into a similar problem using a [fit-PC](http://www.fit-pc.com/web/), so I built a 32-bit build of GRIP.  I can upload it tomorrow night.

ThomasJClark, 2016-01-14:
The 1.0.0 release has been updated to include a 32-bit windows installer (grip-1.0.0-x86.exe)

https://github.com/WPIRoboticsProjects/GRIP/releases/tag/v1.0.0

closed 2016-01-14 by ThomasJClark