WPIRoboticsProjects/GRIP #349: Something to help others that wish to Test this tool with no RoboRio but want to see the NetworkTables results

cpapplefamily, 2016-01-13:
If you run the OnlineViewer tool found in the c:/users/ "current user" /wpilib/tools directory and enter 127.0.0.1 and select start server. Then enter the same IP into grips Settings "NetworkTables Server Address. The Live preview you watch in Grip can display the posted Data live as well. All before you show up to practice and load code into the Robot.

PaulaRudy, 2016-01-13:
Great idea! Thanks again for your suggestions :)

nollchr, 2016-01-14:
Thanks for figuring this out!!

cpapplefamily, 2016-01-14:
I'm trying to set up the Contributing  "setting up build tools" to dig in the code and maybe more help. I Enjoy browsing code and seeing how the insides work. I'm sure I could also contribute to the small minor user comfort items I suggested. I have never use a Git system so this is new ground for me.

tyleraharrison, 2016-01-15:
I am having issues deploying to the RIO. Keeps failing no matter what I do. Tried advanced and regular. Do I need to have GRIP on the RoboRIO to deploy? Any other suggestions?

mdmedoff, 2016-01-30:
Thanks for this post.  I was able to get this working.  However, is it possible to access the Network Tables on the PC from the robot program?  So could you use the sample images in GRIP on your PC and process them in the robot program running on the roboRIO.  If this is possible how do you do this?  How do you tell the robot program where to access the network tables?

cpapplefamily, 2016-01-31:
Use the SmartDashboard. Are you using a USB or IP camera? You can display the video. Also if you wish to view the data on the PC you can use the OnlineViewer as a client enter the IP of the RoboRio 10.TE.AM.20 were TEAM is your ####. As for using the data in the Robot Program Check the Wiki there is sample code there.

mdmedoff, 2016-01-31:
We are using a USB camera.  But to start, I wanted to try using the camera built into my laptop or the sample images with GRIP running on the PC, not on the roborio.  Because I can get GRIP working on the PC and I can see that it is working.  Then I was hoping the robot program running on the RoboRio could read the network tables on the PC.   Is it possible for this to work, or do I have to deploy GRIP to the RoboRio?

cpapplefamily, 2016-01-31:
Set GRIP networkTable to the RoboRio address not 127.0.0.1.GRIP is the Client and will post the data to whatever Network Table address set in its settings. Now use the Laptop Cam or samples. If the robot code is running then the Roborio is serving the table. You can use the OnlineTools now set to 10.TE.AM.20 as "Client" to view the posted Data. As for the video view on the smartdashboard my limited knowledge say NO.

I am not a developer of GRIP but a big fan that has tested and offered small bits of ideas to the development team.

mdmedoff, 2016-02-03:
Thanks again for your help.  We tried what you described above and it worked.  The one other question that I have is when do you set the Online tools to client and when do you set them to server?

cpapplefamily, 2016-02-03:
We're is the network table being hosted? If the roboRio is in the mix then it is the server (you set the online viewer to client ). If testing with no roborio then you must start a server (use online viewer as server) and set it to the local ip 127.0.0.1

AustinShalit, 2016-12-10:
Added to 2017 [ScreenSteps](https://wpilib.screenstepslive.com/s/4485/m/24194/l/479908-reading-array-values-published-by-networktables).

closed 2016-12-10 by AustinShalit