WPIRoboticsProjects/GRIP #350: Feature:  Up/Down Value roll number left of cursor

cpapplefamily, 2016-01-13:
Would be nice to scroll by 1, 10, 100, or .1, .01 ... By sensing the position of the cursor in the text box number could scroll by the tens place.

PaulaRudy, 2016-01-13:
Great idea! I'll mark this down for future discussion/implementation. Thanks for your suggestion!

cpapplefamily, 2016-01-13:
I realy wish I had an idea how to fork and contribute using Github. I'm a new mentor for Team 3244 and a 17 year Electronic Controls Technician working with Allen Bradly PLC and Motion Controls as well as Fanuc Robotics. I am trying to learn as much as possible so I can show the students all the tools at their disposal. Github is a tool I have goals to learn as well.