WPIRoboticsProjects/GRIP #352: Wiki topic: How to deploy to RoboRio

cpapplefamily, 2016-01-13:
I'm struggling on finding how to deploy to the RoboRio or capture the stream from the RoboRio into Grip (on the PC) for processing.

cpapplefamily, 2016-01-13:
I seemed to have make head way by entering roboRIO.####-FRC.local ( ####= Team Number). No errors after hitting Deploy. Now where did the file go? I installed a SFTP browser per the Wpilib setup pages to brows and confirm the myFile.grip is on the Robot.

JLLeitschuh, 2016-01-14:
The core and save file should end up in the lvuser home directory.

cpapplefamily, 2016-01-14:
I did eventually find them there after successfully deploying. I didn't know the confirmation that the deploy worked was the ability to start and stop the app from the deploy screen. I did not get to that step when this question was asked. I also came to the location after digesting the CPP code that points to the file with path as the Java code does not.

tyleraharrison, 2016-01-15:
I am having issues deploying to the RIO. Keeps failing no matter what I do. Tried advanced and regular. Do I need to have GRIP on the RoboRIO to deploy? Any other suggestions?

cpapplefamily, 2016-01-15:
Try to set your rio address in the Setting window not Deploy. When you go back to the Deploy window the Button will be disabled. Simple edit the field but re-enter the same IP. Might work

ThomasJClark, 2016-01-17:
FYI: since there are several open issues and CD posts about problems deploying, I've opened [a new issue](377) where we can collect all of the information in one place.  This should help us track down the underlying problem faster.

closed 2016-01-17 by ThomasJClark

ThomasJClark, 2016-01-22:
[v1.1.0](https://github.com/WPIRoboticsProjects/GRIP/releases/tag/v1.1.0) is out - this should fix most deploying problems