WPIRoboticsProjects/GRIP #353: Deploy error

cpapplefamily, 2016-01-14:
An event subscriber threw an exception
## Stack Trace:
```java
java.lang.IllegalStateException: Tried to stop Webcam 0 but it is already stopped.
	at edu.wpi.grip.core.sources.CameraSource.stop(CameraSource.java:296)
	at edu.wpi.grip.core.sources.CameraSource.onStopPipeline(CameraSource.java:338)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:497)
	at com.google.common.eventbus.EventSubscriber.handleEvent(EventSubscriber.java:74)
	at com.google.common.eventbus.SynchronizedEventSubscriber.handleEvent(SynchronizedEventSubscriber.java:47)
	at com.google.common.eventbus.EventBus.dispatch(EventBus.java:322)
	at com.google.common.eventbus.EventBus.dispatchQueuedEvents(EventBus.java:304)
	at com.google.common.eventbus.EventBus.post(EventBus.java:275)
	at edu.wpi.grip.ui.DeployerController.onDeploy(DeployerController.java:107)
	at edu.wpi.grip.ui.deployment.DeploymentOptionsController.lambda$deploy$21(DeploymentOptionsController.java:81)
	at org.jdeferred.impl.AbstractPromise.triggerDone(AbstractPromise.java:107)
	at org.jdeferred.impl.AbstractPromise.triggerDone(AbstractPromise.java:98)
	at org.jdeferred.impl.DeferredObject.resolve(DeferredObject.java:70)
	at edu.wpi.grip.ui.deployment.FRCAdvancedDeploymentOptionsController.lambda$onDeploy$29(FRCAdvancedDeploymentOptionsController.java:78)
	at org.jdeferred.impl.AbstractPromise.triggerDone(AbstractPromise.java:107)
	at org.jdeferred.impl.AbstractPromise.triggerDone(AbstractPromise.java:98)
	at org.jdeferred.impl.DeferredObject.resolve(DeferredObject.java:70)
	at org.jdeferred.DeferredFutureTask.done(DeferredFutureTask.java:83)
	at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:384)
	at java.util.concurrent.FutureTask.set(FutureTask.java:233)
	at java.util.concurrent.FutureTask.run(FutureTask.java:274)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)

```
## System Info:

Property Name | Property 
 ----- | -----
GRIP Version | 1.0.0
java.version | 1.8.0_66
javafx.version | 8.0.66
os.name | Windows 7
os.version | 6.1
os.arch | amd64

cpapplefamily, 2016-01-14:
It seemed to successfully deploy tho

JLLeitschuh, 2016-01-14:
Ha! I know why this is a thing. Totally forgot about this edge case.
Thanks for reporting.

cpapplefamily, 2016-01-14:
This was my second successful deploy Still having troubles. After the first successful deploy could I simply SFTP the myproject.grip project to /home/lvuser/myproject.grip.? I been on Chief Delphi chatting and now stuck if I get a deploy then I get a play button. I'm trying to get Java to launch Grip. Then Shut it down. 1 to use Video streaming to DriverStation. 2 Could be useful to launch a different Grip Pipeline

JLLeitschuh, 2016-01-14:
The deploy should copy over both the jar and the current save file.
You should just have to hit the **play** button and it should start running on the robo-rio.
Are you not experiencing this?
Can you link to the Chief Delphi conversation?

cpapplefamily, 2016-01-14:
I did experience the play button and verified that NetworkTables is updating with new values. The Next time I tried to deploy it failed a few times before finding the right address/ip syntax/ 

http://www.chiefdelphi.com/forums/showthread.php?t=141097

PR #358 (2016-01-15): Fixes stopped camera exception stop pipeline event
This fixes an issue where the onStopPipeline event
would throw an exception if the camera source was
already stopped.

Closes #353
commit bf0eb353 (2016-01-15): Fixes stopped camera exception stop pipeline event

This fixes an issue where the onStopPipeline event
would throw an exception if the camera source was
already stopped.

Closes #353
closed 2016-01-15 by ThomasJClark