WPIRoboticsProjects/GRIP #354: Start Grip in Java

cpapplefamily, 2016-01-14:
I'm tying the darnedest to get this to work. Here is what I have in my StartGrip.java Command
[CODE]
// RobotBuilder Version: 2.0
//
// This file was generated by RobotBuilder. It contains sections of
// code that are automatically generated and assigned by robotbuilder.
// These sections will be updated in the future when you export to
// Java from RobotBuilder. Do not put any code or make any change in
// the blocks indicating autogenerated code or it will be lost on an
// update. Deleting the comments indicating the section will prevent
// it from being updated in the future.

package org.usfirst.frc3244.ServoTest.commands;

import edu.wpi.first.wpilibj.DriverStation;
import edu.wpi.first.wpilibj.command.Command;
import edu.wpi.first.wpilibj.smartdashboard.SmartDashboard;

import java.io.IOException;

import org.usfirst.frc3244.ServoTest.Robot;

/**
 *
 */
public class StartGrip extends Command {

    // BEGIN AUTOGENERATED CODE, SOURCE=ROBOTBUILDER ID=VARIABLE_DECLARATIONS
 
    // END AUTOGENERATED CODE, SOURCE=ROBOTBUILDER ID=VARIABLE_DECLARATIONS

    // BEGIN AUTOGENERATED CODE, SOURCE=ROBOTBUILDER ID=CONSTRUCTOR
    public StartGrip() {

    // END AUTOGENERATED CODE, SOURCE=ROBOTBUILDER ID=CONSTRUCTOR
        // BEGIN AUTOGENERATED CODE, SOURCE=ROBOTBUILDER ID=VARIABLE_SETTING

        // END AUTOGENERATED CODE, SOURCE=ROBOTBUILDER ID=VARIABLE_SETTING
        // BEGIN AUTOGENERATED CODE, SOURCE=ROBOTBUILDER ID=REQUIRES

    // END AUTOGENERATED CODE, SOURCE=ROBOTBUILDER ID=REQUIRES
    }

    // Called just before this Command runs the first time
    protected void initialize() {
    	String filename = "/home/lvuser/MyFirstFRCDeploy.grip";
    	
    	try {
			Runtime.getRuntime().exec(new String[]{"/usr/local/frc/JRE/bin/java", "-jar", "grip.jar", filename});
			SmartDashboard.putString("Grip", "Started");
			System.out.println("Hello From the StartGrip.java"); //Print to RioLog
			
    	} catch (IOException e) {
			// TODO Auto-generated catch block
    		SmartDashboard.putString("Grip", "Failed");
			e.printStackTrace();
		}
    }

    // Called repeatedly when this Command is scheduled to run
    protected void execute() {
    }

    // Make this return true when this Command no longer needs to run execute()
    protected boolean isFinished() {
        return false;
    }

    // Called once after isFinished returns true
    protected void end() {
    	SmartDashboard.putString("Grip", "End");
    }

    // Called when another command which requires one or more of the same
    // subsystems is scheduled to run
    protected void interrupted() {
    	end();
    }
}
[/CODE]

JLLeitschuh, 2016-01-14:
For future reference the way to do code blocks in markdown is to wrap your code like this

\`\`\`
// Code goes here
\`\`\`

And markdown will automatically format it.

Also, can you explain what problem you are having with this code?

cpapplefamily, 2016-01-14:
Grip is not Launching from with in the Robot Code. The 
...java
SmartDashboard.putString("Grip", "Started");
System.out.println("Hello From the StartGrip.java"); //Print to RioLog
...
is outputting "Started" and "Hello From the StartGrip.java" to tell me the command ran once

cpapplefamily, 2016-01-14:
I tried both ways
"/home/lvuser/grip.jar" in place of "grip.jar"
and 
"/home/lvuser/MyFirstFRCDeploy.grip" in place of "MyFirstFRCDeploy.grip"

AustinShalit, 2016-07-29:
@cpapplefamily I am going to close this issue because it has grown stale.  If you are still having trouble, please take a look at [this Wiki article](https://github.com/WPIRoboticsProjects/GRIP/wiki/Tutorial:-Run-GRIP-from-a-CPP,-Java,-or-LabVIEW-FRC-program).  If you are still having trouble, reopen this issue.

closed 2016-07-29 by AustinShalit