WPIRoboticsProjects/GRIP #356: Do Both the "grip.jar" and "myprodject.grip" need to be deploied each time?

cpapplefamily, 2016-01-14:
Does the jar file change with every saved copy to be deploied so that both files are required to be refreshed each time you wish to try a new vision file? If not and even if so. GRIP could generate the files so a user could manually place the files on the Robot or other slave device

JLLeitschuh, 2016-01-14:
No, the jar is deployed every time in case you download a newer version of GRIP. The jar does not change every time. We want to make sure that you always have the newest version deployed.

If you want to go messing around inside of the program files the only jar you need on the slave device is the grip-core.jar file. The rest are for the UI.

Perhaps we should create a wiki page about this.

cpapplefamily, 2016-01-14:
well then.. until I update the Grip program, FTP the myGripFile.grip will be easiest. The Deploy is hit n miss for me. This is of course if I can get the Robot code to start/stop the app.

JLLeitschuh, 2016-01-14:
Can you explain what you mean when you say that deploy is "hit or miss" for you?

ThomasJClark, 2016-01-14:
> No, the jar is deployed every time in case you download a newer version of GRIP. The jar does not change every time. We want to make sure that you always have the newest version deployed.

Wait, really?  We should probably not deploy the jar if it's already there and it's up to date (maybe check a hash?)

cpapplefamily, 2016-01-14:
I don'T recall all the process I went through a person on chief Delphi said they used roborio-####-frc.local (did not seem to work for me) tried changing  to 10.##.##.20  while in deploy don't  think it worked.  Went to setting put 10.##.##.20 in there, back to deploy had to edit to make button show iT might of worked I don't  recall. There was a lot of keyboard smashing. I will not likely get bake to testing for a day or two.

closed 2017-01-08 by SamCarlberg