WPIRoboticsProjects/GRIP #357: How to be sure Grip is running in the Robot user code?

cpapplefamily, 2016-01-14:
If I want to be sure that grip is running with out errors from my Robot Java or CPP code how can we do this. It maybe nice to know we are getting current data and it has not crashed.

JLLeitschuh, 2016-01-14:
The code you use in java returns a [Process](https://docs.oracle.com/javase/8/docs/api/java/lang/Process.html) object that you can use to determine if the process has died or not.
You may want to put the `waitFor` in another thread and then check the exit status after the process exits.
There are various posts on stackoverflow if you search for "java check if process is running"

AustinShalit, 2016-07-30:
I am going to close this issue because it is stale.  If you still need help, please reopen this issue.

closed 2016-07-30 by AustinShalit